brett8883/DJI_Super-Patcher #7: Phantom 4 Pro not working

GrinFactor, 2019-07-10:
I have at least 3 guys letting me know that the P4P is not working and that the .bin file is not in the super patcher directory when they need to select the .bin file. I've also given them the Cache directory but it still does not work.

They have also commented that the readme file in the P4P cache folder states DJI Spark and not Phantom 4 Pro.

All three have tested and confirmed that altitude, nfz, etc are not working.

brett8883, 2019-07-10:
Yep that’s a bug thanks I’ll have that hot fixed later today

brett8883, 2019-07-10:
This has been fixed remotely.have them try again

GrinFactor, 2019-07-10:
They have tested and it still does not generate or pull the file.

GrinFactor, 2019-07-10:
They also get adb error

GrinFactor, 2019-07-10:
@brett8883 thank you for commenting on the YouTube video! Has it been fixed? There are 4 guys in Facebook that are saying that it's still not working. They tested in the last hour.

brett8883, 2019-07-10:
I have added a stop into the p4p process which will display a message to check for this file. Super-Patcher for p4p works just fine for me this will allow us to figure out what is going on. If a user does not see the debug stop then they are using the wrong version of Super-Patcher or choosing a different aircraft other than p4p. If you could just try it yourself since I know you know how it works to verify the P4P_wm331_Super_Patcher_0306_03.02.44.31_dji_system.bin gets downloaded please. you dont have to connect an aircraft just run the auto_install.bat and choose p4p and when you get to the stop check to see if the P4P_wm331_Super_Patcher_0306_03.02.44.31_dji_system.bin file has been downloaded. This will rule out user error

brett8883, 2019-07-10:
I have multiple people saying that they are not having an issue with the phantom FC downloading or flashing so if a user is still having an issue its probably a local issue and not an issue with Super-Patcher. I have closed the issue but feel free to reopen it or make a new one if you think there is still an issue with Super-Patcher. Thanks!

closed 2019-07-10 by brett8883