brett8883/DJI_Super-Patcher #8: Problem with Windows10 and Windows7/64

Weirdoz, 2019-07-14:
Trying to run the superpatcher on Win10 with VMware fusion on a macbook. When i start the auto install.bat, the window closes immediatly. So i try it on a win7/64 machine, and the autoinstall.bat runs and the download runs.
After that, i could not run the startup.bat of the superpatcher, so i copy all files to the win10 directory. Now the startup.bat runs as it should, and i could do all patches. Both systems runs as admin account. Any idea?
Today i try it via temviewer for a friend on a win10 machine, but we stuck when we should open the adb, it starts, but when the superpatcher runs, the program (adb) stops working (win nag screen). So we stuck now on a clean 1.04.300 fw and cant flash the fc firmware.

brett8883, 2019-07-14:
On the VMware it sounds like it’s not properly downloading the files as you say glad you got that figured out. If you are having an issue with adb it could be due to path traversal when it comes to the remote login your are using but I don’t know that much about the program you are using. Since it sounds like you are Windows proficient I would suggest downloading adb tools for your friend and adding it to his system path so you aren’t relying on the adb tools in the Super-Patcher folder due to path traversal concerns. https://developer.android.com/studio/releases/platform-tools

There is an issue certain users are running into that I am trying to get to the bottom of where adb.exe is continuing to run in the background despite having finished its function. When a new instance of adb.exe is started, say after the aircraft has restarted, a new instance of adb starts running in the background instead of using the already running instance. This eventually leads to adb.exe crashing. Just yesterday I added adb kill-server commands after each instance of adb.exe to make sure only one instance is running at a time. However, if adb was already running in the background then on start up of super Patcher the old folder of Super-Tools-Master cannot be deleted and the new one downloaded because the folder is being used by the system. I am looking at how to best combat this issue as I can’t replicate i on my machine. However, know that when you get “adb.exe has stopped working” error this is an old instance of adb that has crashed not the current one so if you ignore the error and continue the patch will proceed correctly. 

I hadn’t considered the usage of SuperPatcher with a VM and so that might be something I can better accommodate in the future. For now it would be better to use bootcamp to run the Windows machine on a Mac for most users. 

Let me know if any of that helps.

brett8883, 2019-07-14:
Also using task manager to kill background instances of adb.exe before starting Super-Patcher has helped most users that have had this issue. Again this should have been fixed yesterday so might also be worth deleting the “Super-Tools-Master” folder from the Super-Patcher folder to ensure that the new fixes in Super-Tools-Master folder are correctly downloaded at run time.

Weirdoz, 2019-07-15:
Hi, thanks for your very quick response.
The Problem with the VMware i have cleared, because i use both windows versions.
The problem with the win10 and the adb.exe still exist. There is no adb running in background, because this is happening all time, even when we use superpatcher after a complett restart of windows. It looks like adb is running until superpatcher starts to contact to the drone, when we start the the process at superpatcher, adb stops working. At fisrt time we ignore that and start uploading the flightcontroller firmware, but it stucks at the end of uploading, means it does not flash.
But maybe there is also another problem with that win machine, we have no admin rights on it, maybe thats the primary problem.

brett8883, 2019-07-15:
Well yea that’s intentional. I kill all instances of adb that are running in the startup.bat so that when it gets to the adb commands in Super-Patcher it will invoke the copy of adb.exe that is in Super-Patcher. It can’t just be running already to work. The commands have to be able to reference the actual executable file adb.exe. The reason I was recommending adding adb to the system path is that by doing so you are telling Windows “ hey whenever I say adb I’m talking about this file here.” So that way Windows always knows where adb.exe is and doesn’t have to look for it. 

https://www.xda-developers.com/adb-fastboot-any-directory-windows-linux/amp/

This will solve any adb issues immediately.

brett8883, 2019-07-30:
I’ve added some adb kill-server commands and this seems to have fixed this issue I’m gonna close this issue. You are welcome to reopen it if you think the issue still exists and it wasn’t the local machine that was causing the problem

closed 2019-07-30 by brett8883