brett8883/DJI_Super-Patcher #11: PH4ADV Test Patcher Fail

wanghuakangjq, 2019-07-29:
Can't link GPS after cracking and can't take off manually, make sure the remote control is functioning normally.and i don't know how see the log.thanks bor!

brett8883, 2019-07-30:
What flight controller# does the NLD add say you are on?

wanghuakangjq, 2019-08-01:
Thank you for answering my question！ emm is a p4a. iam sure the firmware is right and the NLDapp show the version is right too.i will try again. Thank you again for providing such a good tool，my P4s is success！

Wangk1ng, 2019-08-27:
I also have this problem. P4Adv lost GPS

Wangk1ng, 2019-08-28:
今天我发现这个问题也突然影响了4PRO，这台4PRO是上个月破解的，起初一段时间使用毫无问题，仅仅是这几天突然发生的事情，开机后没有GPS信号，显示ATTI，或者视觉避障模式。这个问题是开启飞机就显示的，而且是每次开机都会显示，即使等待半小时以上，也没有解决问题。前几天新破解了的4Adv破解完就出现了没有GPS信号，刷回官方版本则一切正常。
----------------------------------------------------------------------
The following is the translation provided by the translation software.
-----------------------------------------------------------------------
Today, I found that this problem also suddenly affected 4PRO. This 4PRO was cracked last month. At first, it was no problem to use it. It just happened suddenly in the past few days. There was no GPS signal after booting, showing ATTI or visual obstacle avoidance mode. This problem is shown when the plane is turned on, and every time it is turned on, even if it waits for more than half an hour, it does not solve the problem. A few days ago, when the newly cracked 4Adv was cracked, there was no GPS signal. When it was returned to the official version, everything was normal.

brett8883, 2019-09-02:
House keeping, renamed issue to be more specific

plblacklock, 2019-09-21:
“Sunravers” was apparently able to build a patch to turn off Galileo support and solve the problem. Issue is, there’s no step by step procedure provided for us dummies. I reflashed my P4P V 1 with factory firmware (twice) using assistant 2 and the Galileo function/gps loss/abrupt switch to atti mode still exists. Sounds like the only fix is to swallow pride and send the bird to dji for a full firmware wipe and reload. Any advice before I do so would be immensely appreciated!

brett8883, 2019-10-02:
This issue is for P4Adv let's keep the P4P stuff separate from the P4Adv. P4P patch has been made public at https://github.com/brett8883/DJI_Super-Patcher/issues/17#issuecomment-536314530

brett8883, 2019-10-16:
This next release of Super-Patcher will make Galileo satellite reception optional for all users. It will be an option just like the Smart Battery Mod and Stealth Mod

brett8883, 2019-10-22:
Here is a link to download the Super-Patcher flight controller for [P4Advanced without galileo reception](https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/P4A/2.1/fc_bin/P4adv_Standard_NoGal_03.02.35.66_dji_system.bin) 

Do Super-Patcher but when asked to flash the file with dumldore flash this file instead of the one Super-Patcher tells you to flash. 

when you verify with NLD your flight controller # will be 03.02.35.66 if successful. 

I plan to make Galileo an optional modification in the next version of Super-Patcher

zzb1967736495, 2019-11-22:
我也有这个问题。P4Adv丢失了GPS

cf1234567, 2020-09-07:
> 这个问题是针对P4Adv的，让我们将P4P的内容与P4Adv分开。P4P补丁已在[＃17](https://github.com/brett8883/DJI_Super-Patcher/issues/17#issuecomment-536314530)公开[（评论）](https://github.com/brett8883/DJI_Super-Patcher/issues/17#issuecomment-536314530)
P4A刷机后无法启动电机，也不报错是什么情况

cf1234567, 2020-09-07:
> 这里是一个链接，用于下载[不带伽利略接收的P4Advanced](https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/P4A/2.1/fc_bin/P4adv_Standard_NoGal_03.02.35.66_dji_system.bin)超级修补程序飞行控制器[](https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/P4A/2.1/fc_bin/P4adv_Standard_NoGal_03.02.35.66_dji_system.bin)
> 
> 做Super-Patcher，但是当被要求用dumldore刷新文件时，请刷新该文件，而不是一个Super-Patcher告诉您进行刷新。
> 
> 当您通过NLD验证时，如果成功，您的飞行控制器编号将为03.02.35.66。
> 
> 我计划在下一版的Super-Patcher中将Galileo设为可选修改

P4A在刷入补丁后无法启动电机，飞行器开机自检全部正常，遥控器操作没反应，可以一键起飞，但是不能操作