brett8883/DJI_Super-Patcher #12: Mavic Pro Platinum: Super-Patcher says it succeeded, but NLD shows FC at 03.02.44.07

mtebbens, 2019-07-29:
Bird: Mavic Pro Platinum, FW 01.04.0500 (no internal sd card I suspect).
Computer: MBP running Fusion (Win 10).

Found: https://www.youtube.com/watch?v=2FE5NWEPmmU

Followed the above video step by step .... carefully.
I was able to get Firmware to 01.04.0300 (flashed 2x), then tried Super-Patcher.
Super-Patcher ran smoothly and said it succeeded during the flashing, but in the end NLDApp showed the FC at 03.02.44.07 (not 03.02.44.10). 

Going to try again tomorrow and video the entire process....

Suggestions??

brett8883, 2019-07-30:
Either use a real PC or Mac in bootcamp mode. Super-Patcher doesn’t work well in VMs

mtebbens, 2019-07-30:
Actually it worked great when I used version 1.5 as in the video !
I watched as different results were shown in at least 1 step during the process.

Your latest version 1.7.1 does not play nice with my Mavic Pro Platinum.

I am now at FW 01.04.0300 (flashed 2x), and FC 03.02.44.10 (as shown on NLDApp).

HURRAY!

I believe it was during the first few steps where (with 1.5) I saw:
mkdir failed.....
dummy_verify.sh: 1 file pushed....

I wish I could help you more with what was different, but I really don't want to screw things up now.
If you need me to try/do something to help you, let me know.

Bird: Mavic Pro Platinum (now at FW 01.04.0300 (flashed 2x), and FC 03.02.44.10)
Using: Win10 via Fusion on a MBP (went very smoothly).

brett8883, 2019-07-30:
Very good. I thought it probably had something to do with the ports on the VM but sounds like maybe it just needed a reboot. 

The mkdir failed because it already existed so that step actually did work on the first go around which means it didn’t have anything to do with the ports. Sounds like dummy_verify needed to establish itself before it could bind with dji_verify. 

Glad you got it working!

drak59, 2019-07-30:
i've spent 2 days on it, trying V 1.5 to V 1.7.1 and it doesn't works on my MP1 (like many of my friends) all of us are still in FC 03.02.44.07 and not xx.10 ): maybe one clue...each time during the DUMLdore Enable ADB, the red lights are turning to off but not flashing) then it seems to transfer, no error. I let it down ..really disappointed and sad . But big thanks to the owner

brett8883, 2019-07-30:
@drak59 you must not be doing something correctly. There are plenty of people that have been using successfully. Make sure you have flashed to stock 01.04.0300 twice before starting

mtebbens, 2019-07-30:
> i've spent 2 days on it, trying V 1.5 to V 1.7.1 and it doesn't works on my MP1 (like many of my friends) all of us are still in FC 03.02.44.07 and not xx.10 ): maybe one clue...each time during the [...]

During the specific process of "DUMLdore/Enable ADB" on v1.5 the red lights TURNED OFF on my Mavic Pro Platinum.

brett8883, 2019-07-30:
Yes that’s what is suppose to happen as described by the program during the process.  

“ Please ensure DUMLdore says you are on firmware %stock% and that this is stock version 

Then click "Enable ADB" in DUMLdore and wait for ADB to enable 
The front lights will usually flash, turn on, or **turn off** ”

I am working on the next version of Super Patcher which will have some error handling which will make it more clear as to what goes wrong when something goes wrong. 

People that have had issues with Super-Patcher generally report that trying it on a new computer usually does the trick. I’m not sure why but this seems to solve the issue. 

There’s also a branch 1.7_offline which is a self contained version which doesn’t require internet access. 

https://github.com/brett8883/DJI_Super-Patcher/tree/1.7_OFFLINE

Try this version if you have trouble with 1.7.1

drak59, 2019-07-30:
Thank you all for your support and advices. Unfortunately i did the job like the video ( twice FW update, even if i was already in stock with the  01.04.0300...i 've tested on 2 different computers (one under Seven and other under WIn 10) Each time, i'm going to the end of the process without any errors...Only  the ADB to enable doesn't flash but only turn OFF but the process continue. i Know a lot of people are successful and i'm happy for them and for this great jobs done by you guys...I'm just me and my friends not lucky...all of us are not newbie with computers, that's why we all test everythings and don't let it down in one sec but in 2 days along :-) No worries i'm still your FAN

mtebbens, 2019-07-30:
Even though I had to use SP 1.5 on my bird, am I getting all the goodies you list for 1.7.1 ... Unlocking Height Limits, NFZ limits, Galileo Satellites, fixed fcc and boost ?

In case you add more to newer versions of SP, please see if you can get the Mavic PP working with your latest versions....I want to update again if anything new is added.

"I am working on the next version of Super Patcher which will have some error handling which will make it more clear as to what goes wrong when something goes wrong."

Great... that definitely will help when going through the patch process!
More information during the process will definitely help....

For those trying to get their bird patched using this method;
1. Clean/Fresh install of WIN10 works, I had trouble with WIN7.
2. WIN10 VM on a MBP using Fusion works, that's how I did mine.
3. Follow the steps Slowly and Carefully !!

mtebbens, 2019-07-30:
@brett8883 
Can I Super-Patch my bird AGAIN without causing a problem?

During the first patch JKSON_FCC_MOD offered:

1 - Auto Frequency (Default)
2 - Force Frequency to 2.3G
3 - Force Frequency to 2.5G
4 - Delete this Mod

Since I did not know exactly what each option did exactly, I followed the video and picked 3.
Can someone please briefly explain what each does and why I would want/not want it?

Then POWER:

1 - Default Power
2 - Default > FCC > Boost > Reset. Loop Selector by Remote Controller
3 - Fixed FCC
4 - Fixed FCC & Boost

Again, since I did not know what each option did exactly, I followed the video and picked 4
Now I read that option 4 could cause OVER HEATING via the RF Chips.
Can someone please briefly explain what each does and why I would want/not want it?

This will help me and others reading this decide what they want to do.

Thanks!

brett8883, 2019-07-31:
I have moved a question that was unrelated to this issue to a new issue for the sake of organizing and readability

brett8883, 2019-07-31:
@mtebbens all versions of Super-Patcher work with the Mavic Pro (which is the same as the MPP and alpine white firmware wise) as I have a Mavic Pro and it’s what Super-Patcher gets tested on. 

The flight controllers so far have been the same for all versions of Super-Patcher so yes, using any of their versions ultimately get you to the same point. So yes you still get all the same benefits. 

This will change in the Super-Patcher 2.0 version with the new features that are coming. All the flight controllers will be updated. Having said that, you can still use any version of Super-Patcher to install the new flight controllers you will just have to adjust when it says to flash one of the old (now current) flight controllers to flash the new one instead. And when you verify the flight controller number you’ll just have to verify the flight controller # of the file you flashed instead of the one it says. 

In the 2.0 version there will be multiple options to choose and so this idea of “flight controller numbers will be more important so this will make more sense when the 2.0 version is released. 

Hopefully the new version will just work for you. What’s funny is that the differences between 1.5 and 1.7 are really just visual, 1.7 clears the screen for each step instead of having a rolling list of instructions. The core programming is virtually identical.

drak59, 2019-07-31:
End of bad story for me :-) i was successful and wanted to report my case. First try was on Win 7 (failed), second try on Win 10 (old) Failed...The good one was on WIn 10 Last version and it took me 3 minutes to be happy. So it was because of the computer OS but why ?? ( maybe some last apps, plugins, updates or CPU is important...dunno...One clue for others (if your patched bin flash is very fast (even with 100% done at the end) so it is not transferred...The patched bin 1.5mo takes some times (around 2 minutes) Thanks again to the big brain Brett8883

brett8883, 2019-07-31:
@drak59 very good! Thanks for reporting back!

I am somewhat flabbergasted as to why some people have such a difficult time on one PC but it works perfectly on others... it’s not clear to me what the difference could be. Anyway I hope you are able to help your friends that were having trouble as well.

brett8883, 2019-07-31:
I’ve marked this issue as resolved which means the issue was able to be worked around by the original poster but underlying issue may still exist so the issue will remain open

cderoche, 2019-09-08:
I'm having a similar issue with a friend's Mavic. NLDApp showed the FC at 03.02.44.07, and something weird happens now each time I redo all the steps (tried 5 times no success), it skips past the part where you're suppose to flash the custom firmware and goes right to the JCKSON boost option. And suggestions? I did this procedure successfully with my own Mavic following digdat0's video...
https://www.youtube.com/watch?v=rFrQ52m8bos

cderoche, 2019-09-08:
Also noticed SD Card error after one of the attempts, on the tablet after trying to test fly the drone. Wonder if the internal SD Card is now the issue, making the super patcher program skip that last step to flash the new software package over...

Weirdoz, 2019-09-08:
How about the superpatcher error logs?

cderoche, 2019-09-08:
> How about the superpatcher error logs?

I didn't notice anything obvious, although some of the log I'm sure is above my pay grade. I'll try again when I get home from work and post the log here. I'm thinking the internal SD Card might be corrupt and maybe removing it may help, since it's not needed anymore. I had to add the SD Card to partially mod it using the older method since it's a "newer" Mavic Pro.

cderoche, 2019-09-08:
OK just tried again, it skipped the second enable ADB step and the screen that prompts you to flash the custom firmware, goes straight to JKSON FCC MOD. I attached the error log.
[log.txt](https://github.com/brett8883/DJI_Super-Patcher/files/3588075/log.txt)

cderoche, 2019-09-08:
START SUPER_PATCHER 
Sun 09/08/2019_16:07:39.43 
I have verified internet access to GitHub 
16:07:41.82 
 
Download NLD app 
--2019-09-08 16:07:45--  https://nolimitdronez.com/downloads/nldapp.zip
Resolving nolimitdronez.com (nolimitdronez.com)... 85.184.161.145
Connecting to nolimitdronez.com (nolimitdronez.com)|85.184.161.145|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2072561 (2.0M) [application/x-zip-compressed]
Saving to: 'nldapp.zip'

     0K .......... .......... .......... .......... ..........  2%  172K 11s
    50K .......... .......... .......... .......... ..........  4% 3.07M 6s
   100K .......... .......... .......... .......... ..........  7%  347K 6s
   150K .......... .......... .......... .......... ..........  9% 3.58M 4s
   200K .......... .......... .......... .......... .......... 12%  394K 4s
   250K .......... .......... .......... .......... .......... 14% 3.66M 3s
   300K .......... .......... .......... .......... .......... 17% 3.72M 3s
   350K .......... .......... .......... .......... .......... 19% 3.66M 3s
   400K .......... .......... .......... .......... .......... 22%  431K 3s
   450K .......... .......... .......... .......... .......... 24% 3.79M 2s
   500K .......... .......... .......... .......... .......... 27% 3.71M 2s
   550K .......... .......... .......... .......... .......... 29% 4.03M 2s
   600K .......... .......... .......... .......... .......... 32% 3.39M 2s
   650K .......... .......... .......... .......... .......... 34%  530K 2s
   700K .......... .......... .......... .......... .......... 37% 3.85M 2s
   750K .......... .......... .......... .......... .......... 39% 3.94M 1s
   800K .......... .......... .......... .......... .......... 41% 3.54M 1s
   850K .......... .......... .......... .......... .......... 44% 3.97M 1s
   900K .......... .......... .......... .......... .......... 46%  523K 1s
   950K .......... .......... .......... .......... .......... 49% 3.73M 1s
  1000K .......... .......... .......... .......... .......... 51% 3.15M 1s
  1050K .......... .......... .......... .......... .......... 54% 3.92M 1s
  1100K .......... .......... .......... .......... .......... 56% 3.74M 1s
  1150K .......... .......... .......... .......... .......... 59% 2.83M 1s
  1200K .......... .......... .......... .......... .......... 61%  606K 1s
  1250K .......... .......... .......... .......... .......... 64% 3.90M 1s
  1300K .......... .......... .......... .......... .......... 66% 4.13M 1s
  1350K .......... .......... .......... .......... .......... 69% 3.73M 1s
  1400K .......... .......... .......... .......... .......... 71% 3.51M 0s
  1450K .......... .......... .......... .......... .......... 74% 3.82M 0s
  1500K .......... .......... .......... .......... .......... 76% 4.00M 0s
  1550K .......... .......... .......... .......... .......... 79% 4.20M 0s
  1600K .......... .......... .......... .......... .......... 81% 2.97M 0s
  1650K .......... .......... .......... .......... .......... 83% 3.44M 0s
  1700K .......... .......... .......... .......... .......... 86% 1.97M 0s
  1750K .......... .......... .......... .......... .......... 88% 3.93M 0s
  1800K .......... .......... .......... .......... .......... 91% 3.26M 0s
  1850K .......... .......... .......... .......... .......... 93% 3.29M 0s
  1900K .......... .......... .......... .......... .......... 96% 3.78M 0s
  1950K .......... .......... .......... .......... .......... 98% 4.10M 0s
  2000K .......... .......... ...                             100% 3.32M=1.4s

2019-09-08 16:07:48 (1.40 MB/s) - 'nldapp.zip' saved [2072561/2072561]

starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
adb shell mkdir /vendor/bin 
adb push dummy_verify.sh /vendor/bin/ 
 
dummy1 engaged based on AC type is MavicPro 1 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
adb shell mount -o remount,ro /vendor 
dummy_verify step complete 
Starting bind step... 
bind1 engaged based on AC type is MavicPro, verify type is 1 
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify 
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found
Start flash step 
dummy_bind complete 
Start jkson 
User requests jkson fcc mod, starting now 
jkson mod complete 
going to _verify 
_verify.cmd starting 
User Indicated patch not successful. Showing log file

cderoche, 2019-09-09:
OK tried it again after pulling out the added internal micro SD card, still didn't work. Still skipping second ADB step and entire flash new firmware step.
[log.txt](https://github.com/brett8883/DJI_Super-Patcher/files/3590715/log.txt)

cderoche, 2019-09-09:
OK I _think_ I finally got it, finally shows correct flight controller number. Now to test fly it.

For some reason, super patcher skipped a couple of the prompts where it asks you to enable ADB, the DUMLDORE app popped up each time though. I rewatched the digdat0 guide video to catch the parts where to enable ADB, and that seemed to be the whole issue, it didn't skip the part asking to flash the new mod firmware this time either....

brett8883, 2019-09-10:
Sometimes if you are over zealous it can be possible to hit “any key” twice which will

cderoche, 2019-09-10:
I thought it could have been that too at first, but I was very methodical and deliberate after the first time it messed up, and probably ran that guy's Mavic through all the steps I'll bet 20 times, not really sure what went on with his, the program did perfectly for mine.

brett8883, 2019-10-29:
This issue should be fixed. I swapped out the start %dumldore% commands for call %dumldore% on 2.0 and above so the program will not continue until DUMLdore has been closed

closed 2019-10-29 by brett8883