brett8883/DJI_Super-Patcher #13: @brett8883

brett8883, 2019-07-30:
@brett8883 
Can I Super-Patch my bird AGAIN without causing a problem?

During the first patch JKSON_FCC_MOD offered:

1 - Auto Frequency (Default)
2 - Force Frequency to 2.3G
3 - Force Frequency to 2.5G
4 - Delete this Mod

Since I did not know exactly what each option did exactly, I followed the video and picked 3.
Can someone please briefly explain what each does and why I would want/not want it?

Then POWER:

1 - Default Power
2 - Default > FCC > Boost > Reset. Loop Selector by Remote Controller
3 - Fixed FCC
4 - Fixed FCC & Boost

Again, since I did not know what each option did exactly, I followed the video and picked 4
Now I read that option 4 could cause OVER HEATING via the RF Chips.
Can someone please briefly explain what each does and why I would want/not want it?

This will help me and others reading this decide what they want to do.

Thanks!

_Originally posted by @mtebbens in https://github.com/brett8883/DJI_Super-Patcher/issues/12#issuecomment-516623998_

brett8883, 2019-07-30:
@mtebbens your question was unrelated to the issue that you originally posted it in. I have moved your question here and deleted your original post. 

You would be better off just doing the stand alone version of jkson fcc mod instead of repatching, though no, you wouldn’t mess anything up it just isn’t necessary. 

I can only speak to the frequencies and power form my own experience because I am not the creator of jkson_fcc_mod but I have never had an issue with the fixed boost + FCC and never heard of someone who has. 

I think choosing “auto” (#1) in the frequency selection will be best in 99.99% of cases. There was a moment I thought about not offering the other options for frequency in Super-Patcher but decided it wasn’t my right to decide what was best for people. It’s possible I may restrict the options in future versions just due to the amount of confusion about them but you can always do the stand alone version here https://github.com/jkson5/jkson_fcc_mod

xitanaf, 2019-08-12:
Hi brett, I tested the Jkson mod on Spark.
"busybox ping -c 1 -w 1800 192.168.41.2" Spark does not seem to support it
"dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000" Spark set up FCC at start/restart, but apk Go4 display "prompt to change region OK / Cancel"

brett8883, 2019-08-12:
@xitanaf jkson fcc mod does not have documented support for Spark but from jkson5’s response to some of the questions asked of him it sounds like there is at least an outside possibility it could work. If you’ve been able to find a way to get it to force fcc mode on Spark that’s a big accomplishment. I don’t have a spark and I already live in FCC in a fcc country so I can’t test it. 

Give us an update on how it goes and if you notice better reception.

xitanaf, 2019-08-13:
OK. I have 2 mods. Force FCC and Force FCC+TXPOWER
Force FCC is standard, any start/restart aircraft, the dron automatic set FCC
TXPOWER is as Boost, the distance is a little better, about +20%, but signal stability is great. The image is fixed until the end of the range

Here is extract from file check_1860_state.sh
Force FCC:
#!/system/bin/sh 
/system/bin/check_1860_state.sh& 
while : 
do 
dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000 
break 
done

Force FCC+TXPOWER:
#!/system/bin/sh 
/system/bin/check_1860_state.sh& 
while : 
do 
dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000 
sleep 10 
iw phy phy0 set txpower limit 3500 
break 
done

brett8883, 2019-08-13:
Nice! I’m gonna have  some people do some tests for me and if it pans out I’ll add it to Super Patcher and credit you for your work! 

Great work!

brett8883, 2019-08-13:
@BorisPlintovic do you have any insight on what all tat is actually doing? And how did you find out what to do? I’ll admit while I understand what it does I don’t understand the mechanics of how it works. 

Any idea if this would work on other drones?

brett8883, 2019-08-13:
Oh wait😲. I think I might get it. So you put Spark into FCC mode and then pulled the Check_1860.sh to check the settings and then you can call the modified check_1860.sh at start up?

BorisPlintovic, 2019-08-13:
Tommorow i send you .bat file for Spark.

brett8883, 2019-08-14:
> Tommorow i send you .bat file for Spark.

Thanks But is that how you figured out what settings to put in to force it into fcc? I’m really just curious how that was done. Maybe we can get it working for other aircraft too.

BorisPlintovic, 2019-08-14:
For over a year I used the command "iw reg set US" to switch to the FCC
A week ago I tried to use the jkson5 command for MavicPro "dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000" and it work just as well.
I think the jkson5 FCC command will work equally well on all drones.
Here is the .bat file for Spark:
https://drive.google.com/open?id=1vkLXswF3P547JGORAlZMrrDQd9fCj1gs

ausdim, 2019-08-14:
Try that on my spark but didnt turn to fcc.
Btw bird allready patced with latest super-patcher.

Thanks

BorisPlintovic, 2019-08-14:
I tested it on
FW AC 0900, 1000, 0900 SuperPatcher
FW RC 0400
Go4 4.3.24
After installing fcc mod, apk will automatically report (popup window) a different region and asks to restart wifi (OK / Cancel). Click to Cancel and FCC to stay

mosheas, 2019-08-15:
Hi
Did you get a greater distance and stable image quality by the end of the reception area?
At what frequency did you work 5.8 or 2.4?

brett8883, 2019-08-15:
> I tested it on
> FW AC 0900, 1000, 0900 SuperPatcher
> FW RC 0400
> Go4 4.3.24
> After installing fcc mod, apk will automatically report (popup window) a different region and asks to restart wifi (OK / Cancel). Click to Cancel and FCC to stay

Thanks @BorisPlintovic. I’ve given it to several beta testers to try out so I’ll keep you updated on the feedback. Everybody is excited about the prospect of a Spark FCC mod.

mosheas, 2019-08-15:
Thanks so much .. Definitely exciting to hear good news (:

BorisPlintovic, 2019-08-15:
Fly in 2.4
I fly out of city and 2.4 have better penetration od terrain
In this mod, 2.4 and antenna 4hawks SR is my distance record 6105m
https://youtu.be/zeXvTDzU-Ko

ausdim, 2019-08-16:
Tried on 0900 but cant go to fcc.
I try also to reflash stock 0900 firmware and super patcher and again fccmod with the same results again.
Btw all seems ok and the .sh file is at vendor/bin folder.

Weird.
Also as i see your versions the only diff is that my remote is not 0400 cant remeber exactly cause im at work.
Also where you found 4.3.24 go 4 apk?

Many thanks for your work.

mosheas, 2019-08-16:
Disappointing me to hear that it didn't work...

ausdim, 2019-08-16:
Dont want to disappoint anyone maybe is something at my side only. I wish...
Tried both the fccmod bat file and also tried to manually write the commands throw adb shell but same results.
Fcc comes only if i use fake gps.

Also mkdir vendor/bin gives an error that folder allready exists maybe a "-p" fix that..

Any way im here if something new popup to test it.

Thanks

BorisPlintovic, 2019-08-16:
then the problem is in rc firmware 600

ausdim, 2019-08-16:
Thats maybe the problem...
I try to downgrade and see the results.
What the diff with that two rc firm? Otg stuff? Soory but cant search now @work

BorisPlintovic, 2019-08-16:
One idea, you can try rewrite "dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000“  to "iw reg set US" in .bat file

ausdim, 2019-08-16:
Allready tried "iw reg set US" via adb shell but no go...
But i will try again when i go home.

BorisPlintovic, 2019-08-16:
interestingly, I also tried "dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000" manually in ADB and did not get any error

ausdim, 2019-08-16:
Isnt "error" but something else.. bont know what.. i will post it back when i am @home

brett8883, 2019-08-16:
@BorisPlintovic with the limited feedback I’ve got back so far things are looking good. The one beta tester who has reported back is very impressed. He says he got over a mile in a highly congested area. I’d like to get some more feedback from a few other people but wanted to give you the update

ausdim, 2019-08-16:
Thats great news... 
What rc firmware user is on?
Thanks

ausdim, 2019-08-17:
Thats the message i got when i try to manually put the code via adb shell.

> root@wm100_dz_ap0001_v5:/vendor/bin # dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000
> Send message, from 1e00, to 0900, msg_id: 00090027, attrib: 00000040, len: 11, data:
>   00 02 48 00 ff ff 02 00 00 00 00
> Resp message, len = 1, data:
>   e0
> root@wm100_dz_ap0001_v5:/vendor/bin #

Also just downgrade my rc to 0400 and reflash stock 0900 ac firmware and restore factory defaults via assistant2.

Thanks

brett8883, 2019-08-17:
> Thats the message i got when i try to manually put the code via adb shell.
> 
> > root@wm100_dz_ap0001_v5:/vendor/bin # dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000
> > Send message, from 1e00, to 0900, msg_id: [00090027](tel:00090027), attrib: [00000040](tel:00000040), len: 11, data:
> > [00 02 48 00](tel:00 02 48 00) ff ff [02 00 00 00 00](tel:02 00 00 00 00)
> > Resp message, len = 1, data:
> > e0
> > root@wm100_dz_ap0001_v5:/vendor/bin #
> 
> Also just downgrade my rc to 0400 and reflash stock 0900 ac firmware and restore factory defaults via assistant2 but the bird dont want to go to fcc.
> 
> Thanks

Hold up you are just typing the commands into the shell at vendor/bin?

ausdim, 2019-08-17:
That time yes but tried also from root.
Spent a lot of time to that situation... dont know maybe the fault is somewere else..
How you have reports that is working and for me isnt works? Make me think...
Thanks

mosheas, 2019-08-17:
@BorisPlintovicoes 
Does the spark file.bat, you've moved her should to work well, or do i need to change any parameters?, thanks.

brett8883, 2019-08-17:
> That time yes but tried also from root.
> Spent a lot of time to that situation... dont know maybe the fault is somewere else..
> How you have reports that is working and for me isnt works? Make me think...
> Thanks

I just have the one tester who has reported back so far but he’s usually not easily impressed but he’s over the moon lol. Says he moved to a lower interference area and can’t out fly the signal range before needing to return due to low battery.

brett8883, 2019-08-17:
> @BorisPlintovicoes
> Does the spark file.bat, you've moved her should to work well, or do i need to change any parameters?, thanks.

There’s nothing additional to the bat file that is needed.

mosheas, 2019-08-18:
I tested it today ver - RC 400 and AC 900, app ver 4.3.20, I have pached by the File.bat for Spark, so In testing no change to fcc, I was not notified of zone change, basically nothing happened. I restarted several times, and nothing changed

BorisPlintovic, 2019-08-18:
channel chart in apk may be wrong
Takeoff, landing and after check log file in dji/dji.go.v4/log/cache/areacode, if you see US

mosheas, 2019-08-18:
In the testing the distance was 300 meter on open area then I lose connection,It's exactly the same place same distance before I did a fcc patch

ausdim, 2019-08-18:
There is no any file with name areacode in:
dji/dji.go.v4/log/cache/areacode
Exept if you mean destination:
dji/dji.go.v4/log/cache/areacode/ 
That have only encrypted log files...

BorisPlintovic, 2019-08-18:
./DJI/dji.go.v4/LOG/CACHE/areacode/*.txt
file is NOT encrypted!
my last LOG:

2019-08-15 10:24:12:832 E: init not ru ac=US
2019-08-15 10:24:12:839 E: init success
2019-08-15 10:24:12:847 E: --------DeviceAC getRcVer:mRcVer = 
2019-08-15 10:24:12:857 E: --------DeviceAC getRcVer:onSuccess，mRcVer = 01.00.00.84
2019-08-15 10:24:12:862 E: --------DeviceAC closeJp57
2019-08-15 10:24:12:958 E: --------DeviceAC getCcFromRc onSuccess， ccrc8583
2019-08-15 10:24:12:962 E: --------DeviceAC getCcFromRc onSuccess， mCcFromRc=US
2019-08-15 10:24:13:187 E: --------DeviceAC activity event
2019-08-15 10:24:14:803 E: --------DeviceAC activity event
2019-08-15 10:24:19:327 E: --------DeviceAC activity event
2019-08-15 10:24:20:879 E: --------DeviceAC activity event
2019-08-15 10:26:19:799 E: --------DeviceAC device connect 
2019-08-15 10:26:19:809 E: --------DeviceAC device connect ac is not empty
2019-08-15 10:26:19:892 E: --------DeviceAC activate state != 0
2019-08-15 10:26:20:506 E: --------DeviceAC getCcFromSky onSuccess， mCcFromSky=US
2019-08-15 10:26:20:712 E: --------DeviceAC 【国家码设置（set）】结果onSuccess，:model=dji.midware.data.model.P3.DataWiFiSetWiFiCountryCode@32551ad
2019-08-15 10:28:49:112 E: --------DeviceAC device connect 
2019-08-15 10:28:49:125 E: --------DeviceAC device connect ac is not empty
2019-08-15 10:28:49:312 E: --------DeviceAC activate state != 0
2019-08-15 10:28:51:037 E: --------DeviceAC getCcFromSky onSuccess， mCcFromSky=US
2019-08-15 10:28:51:424 E: --------DeviceAC 【国家码设置（set）】结果onSuccess，:model=dji.midware.data.model.P3.DataWiFiSetWiFiCountryCode@32551ad
2019-08-15 10:29:27:928 E: --------DeviceAC activity event

everywhere to be "US" iam in FCC mod

ausdim, 2019-08-18:
Weird mine is encrypted as i say.

> GmcONIA9PFzHw1gpPLLPoqev3ytAH5KO3S/CJpMsylg= 61apw83Vyxd9roBFy8cHdv7/D4Gxan5sDq/fe8BtinY= CU/fGvftSa5DpJjqgBSF4QzEq1nlRIHhqYIIOuY9dy4= lW6jm5YHwGfcySHj8uzkvw== B2NFbKjer9oxfsRS7VsAB7r3afdPn21bj5YZzWWh9vY= cwgQW1UgxyVfBeSrdwXHsJBITrChvfKPjO2Fzj7Ya/Q= mjYT3jHnJl6YGWKgxfvdBA== mAzeFbkWBGyzJRgEWJSZYQNn44wFaguEUnqkvHVinQw= aBDeHnsfPHLfjb30LOifVTxUi8Vi+K8bS/xr77kygfn4M+ng0zgV7MfM0boILdkc 2lfNnhNY+w/3xMaMUxJbE0YFT/wvMlSDz8Vrd5FEs18= 9GH7Jv34mibPZ3DlZRcMlHB74EpQClQBpTuH6mQ5TAOQcUjrG1NNNFiT0txdTWSH jx+tHKi664Bi0FZXsdhGAcHyuJCKrTNJzmRugiQkupKAA01iHE4Qln1MJ5oj3zpo UXyheC9UQbb/5jqiUQ1OdGlalwyeQ5GUeulZ8KHePsQ

4.3.20 from playstore.
sammy s7.

mosheas, 2019-08-19:
My log file on testing 

2019-08-17 09:25:35:373 E: first get p loc
2019-08-17 09:25:35:409 E: -------start load city list--------
2019-08-17 09:25:37:019 E: city list load finish
2019-08-17 09:25:37:183 E: build kdtree finish count=138456
2019-08-17 09:25:37:232 E: find near city=PS, index=99
2019-08-17 09:25:37:259 E: find near city=IL, index=98
2019-08-17 09:26:02:087 E: --------DeviceAC activity event
2019-08-17 09:27:27:170 E: init not ru ac=null
2019-08-17 09:27:27:180 E: init success
2019-08-17 09:27:28:761 E: init not ru ac=null
2019-08-17 09:27:28:769 E: init success
2019-08-17 09:27:28:781 E: --------DeviceAC getRcVer:mRcVer = 
2019-08-17 09:27:28:855 E: --------DeviceAC closeJp57
2019-08-17 09:27:28:855 E: --------DeviceAC getCcFromRc onFailure：ccodeNOCONNECTccrc4549
2019-08-17 09:27:28:860 E: --------DeviceAC getCcFromRc onFailure：ccode=NOCONNECTmCcFromRc=-1
2019-08-17 09:27:29:233 E: --------DeviceAC activity event
2019-08-17 09:27:29:468 E: first get p loc
2019-08-17 09:27:29:485 E: -------start load city list--------
2019-08-17 09:27:30:286 E: --------DeviceAC activity event
2019-08-17 09:27:30:288 E: city list load finish
2019-08-17 09:27:30:337 E: on p loc change dis=5111414.5
2019-08-17 09:27:30:352 E: build kdtree finish count=138456
2019-08-17 09:27:30:366 E: find near city=IL, index=99
2019-08-17 09:27:30:376 E: --------DeviceAC 拉取激活状态失败 ccode=NOCONNECT
2019-08-17 09:27:30:464 E: get ac from net success t={"extra_msg":["","","84.228.16.246"],"msg":"success","result":{"country_code":"IL","country_name":"以色列"},"status":0}
2019-08-17 09:27:48:901 E: --------DeviceAC activity event
2019-08-17 09:28:00:949 E: --------DeviceAC activity event

brett8883, 2019-08-19:
And you guys having trouble you are using the batch file Boris uploaded above correct? You haven’t modified it in any way? You aren’t just adding his command to some other script correct?

ausdim, 2019-08-19:
Yes i use multiple times the bat file untouched.
Also restock my ac with 0900 and reflash super pather and again the bat file with no fcc again.
The only think havent do is to try fcc mod bat at stock firmware i mean with no any mods on it. (Super-patcher)

Thanks

mosheas, 2019-08-19:
Yes, I using only with the batch file Boris uploaded, I didn't used with super patcher or other scripts command in any time.

BorisPlintovic, 2019-08-19:
I do not understand where the problem may be. :(
I have 2 sparks, one for testing, the other for flying, it works on both.

mosheas, 2019-08-19:
Boris... Maybe because they are constantly on fcc so you have no problem, try changing to CE and then reboot and check if you have a message to change zone

BorisPlintovic, 2019-08-19:
I try change command chmod and delete busybox.
Try new version https://drive.google.com/open?id=1vybqWPsern6zRj30243pJvcDoNY2T4R5

BorisPlintovic, 2019-08-19:
> Boris ... Možno preto, že sú neustále vo fcc, takže nemáte žiadny problém, skúste prejsť na CE a potom reštartovať počítač a skontrolovať, či máte správu o zmene zóny.

So I did. Use apk to switch to ce, check the log and graph in apk (13ch). He then applied the mod and received a warning from apk that the drone was in the FCC

mosheas, 2019-08-19:
I did it with fcc spark_v2.0,no change it's doesn't work.   ):

mosheas, 2019-08-19:
Boris please see PrtScrn of process of patch ver_2.0,  Do you think this is correct? 

https://drive.google.com/file/d/121KejhSJ51gVsYH0TY-kNWBSEWtaL_6Q/view?usp=drivesdk

BorisPlintovic, 2019-08-19:
its OK. If file or folder exist, is automatic rewrite.

mosheas, 2019-08-20:
Boris
I still don't understand how it works for you, and it doesn't work for me, what's the difference in the process?

BorisPlintovic, 2019-08-20:
1. Drone an rc is in CE (13ch)
2. Turn of RC and apk
3. Conect the drone to PC and aply fcc mod
4. Restart drone, turn on RC and wait for green LED
5. Conect OTG to phone, to rc (or with wifi its same)
6. When the apk start, take me popup, i click to Cancel and Go
7. 11ch, drone is in FCC
video: https://drive.google.com/open?id=1beBuEjK5Kl2jOB0mpV9lsUaKSPlvXpfN

mosheas, 2019-08-20:
Well, it does work for you, I'll try again later, do you think the boost works for you too?

BorisPlintovic, 2019-08-20:
maybe it's step by step. first pair the drone and controller and apk last

mosheas, 2019-08-21:
HI Boris.. 
I did exactly according to your video, and it doesn't work for me, .. Some questions ... Do you use the manufacturer's official firmware(900) ? Do you think my problem it is related to the geographical location?  Because in my country it's only allowed to fly at a 2.4ghz frequency limited by the dji standard CE.. Dji closes a frequency of 5.8ghz, is it possible that it has an effect?  so maybe we should change the patch?

Weirdoz, 2019-08-21:
Hi Mosheas
The most issues are based in the wrong windows version. I tried a lot, and the only one who is always working without any problems is a clean win10pro, 64bit. Any other like win10/32, win7/64 or 32bit are not working really good.
So try it on the right win version, believe me it works.

mosheas, 2019-08-21:
I definitely use with the windows 10Pro /64bit, I have no problem with the process, the process is well executed, the problem that it doesn't work for me.

BorisPlintovic, 2019-08-21:
> HI Boris..
> I did exactly according to your video, and it doesn't work for me, .. Some questions ... Do you use the manufacturer's official firmware(900) ? Do you think my problem it is related to [...]

In Slovakia we can use 2.4 and 5.8. I fly 2.4 because of having better penetration in terrain.
I tested on the official 0900, 1000 and 0900FC-patcher, making me work all versions. Do you think there is a special version of FW for IL? You can try flash official fw using DUMLDore, first download fw via DDD ...

BorisPlintovic, 2019-08-21:
Today get feedback from one user, his apk still display 13ch but he fly away >1200m, that means fcc be active.

brett8883, 2019-08-21:
Hey @BorisPlintovic. Was it by design that you have the sleep command as part of the loop?

On standard jkson fcc the check1860.sh is:

``#!/system/bin/sh

/system/bin/check_1860_state.sh& 
busybox ping -c 1 -w 1800 192.168.41.2 
sleep 5 
while : 
do 
dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000 
dji_mb_ctrl -S test -R local -g 9 -s 9 -c 3c 
break 
done`

Between the "while" and "break" is a loop. Notice that the sleep 5 is outside the loop

Your script  is:

`#!/system/bin/sh 
/system/bin/check_1860_state.sh& 
while : 
do 
dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000 
sleep 10 
iw phy phy0 set txpower limit 3500 
break 
done` 

I notice here the sleep 10 is inside the loop. 

Now I dont know if that makes any difference in the first place but just curious if that was on purpose.

The dji_mb_ctrl -S test -R local -g 9 -s 9 -c 27 00024800FFFF0200000000 is what get looped I believe

[28 more replies]