brett8883/DJI_Super-Patcher #14: SP v1.5: Forced descent and landing for low battery at high ALT and I was not at low battery level.

mtebbens, 2019-08-01:
First test flights today using SP v1.5 on my Mavic Platinum ... Very Happy!

Height Limits: My bird flew 1 mile high today ... Works!
NFZ Limits: Not in an area to test this yet ... Will Update
Galileo Satellites: DJI GO showed 20+ Sats ... I assume this works.
I'm at Fixed FCC and Boost ... Guess it works, didn't lose signal.

Now for something I did not expect:
Somewhere between 5000ft - 6000ft with Bat Level at approx. 50%, the DJI Go app displayed Low Bat Warning and then the bird started a Forced Decent and Landing at non-sport mode speed. I could not switch back to Sport Mode to descend quickly. If I was to high the slow decent could have used up all the bat and killed my bird.

Is this an issue with Super-Patcher, or just a parameter I have to adjust... or something else?

brett8883, 2019-08-02:
This is not an issue with Super-Patcher. This is the expected behavior for A MPP with the battery settings set to "DJI-Intelligent battery" mode. 

See great video by digdat0 on how to set the parameters to disable this. https://www.youtube.com/watch?v=BnNo3lWoXkQ

 In Super-Patcher 2.0 there will be a big update on this with a new mod that tackles this particular issue. Please keep checking back for this new feature should be in the next couple of weeks time permitting

brett8883, 2019-08-02:
This issue has been tagged a "question because it is not an issue and is quite literally a question but the  information is relevant so I will keep this issue open for now

brett8883, 2019-08-02:
I’ve added this to the readme