brett8883/DJI_Super-Patcher #16: Inspire 2 V01.02.0300

xstazio, 2019-08-13:
Any chance of using the latest firmware on I2?

brett8883, 2019-08-14:
> Any chance of using the latest firmware on I2?

Probably not to be honest. Do you really use that Multilink thing? That’s the only update in V01.02.0300 even DJI says this 

Date: 2019.01.28
Aircraft Firmware: v01.02.0300
Remote Controller Firmware: v01.01.0060
DJI GO 4 app: iOS v4.3.10 or above, Android v4.3.10 or above

What’s New?

The Inspire 2 remote controller is now compatible with Multilink and can be used as a master or slave controller in the master-slave controller network.

Notes:

Make sure the Multilink is not attached to the remote controller during the firmware update.

If the firmware update fails, restart the remote controller and retry.

If you do not intend to use Multilink, then you are not required to update the firmware for the Inspire 2.

xstazio, 2019-08-14:
I'm anxious about the battery problem:
“New firmware updates are now available for users of TB50 and TB55 batteries to enhance the battery management system and optimize power supply during flight, if you have the corresponding battery, please kindly update the firmware to the latest version, if you have multiple batteries, please run this firmware update on all of them.”

brett8883, 2019-08-15:
> I'm anxious about the battery problem:
> “New firmware updates are now available for users of TB50 and TB55 batteries to enhance the battery management system and optimize power supply during flight, [...]

Are you sure the battery update was in V01.02.0300? I don’t have an Insipre and I don’t pay as much attention to the updates for Inspire 2 as say the Mavic series but just looking at the release notes it looks like there was a battery firmware update in 01.02.0200 but I don’t see one in the release notes for V01.02.0300 unless I’m reading it wrong. At any rate no I don’t have plans to make a patch for the newer firmware. 

If somebody else wants to do it I will gladly add it to Super-Patcher and add them as a collaborator. As I don’t have an Inspire 2 it’s basically impossible to test and I wouldn’t ask somebody to risk it on that expensive of a bird that isn’t mine.

brett8883, 2019-09-15:
This feature is being worked on. With any luck this should be incorporated soon.