brett8883/DJI_Super-Patcher #17: Phantom 4 Pro lost GPS signal.

sunravers, 2019-08-13:
Hello. Thanks for your work! In no-fly zones, the drone can fly, checked.

Already 2 times successfully installed the super patch, and when flying anywhere the same problem arises. Sometimes, for no apparent reason, the drone switches to atti mode for a few seconds (maybe half a minute) and shows a very low GPS signal level, although at that moment it catches about 20 satellites and flies where there are no problems with standard firmware (1.05.0600). At this time, the indicators of the drone light up in red without blinking. After installing the super patch, I calibrated IMU and compass, it does not help. It becomes scary to fly, especially in the wind. I consider this problem to be serious. Maybe the firmware does not know how to process the signal from Galileo satellites?

brett8883, 2019-08-14:
> Hello. Thanks for your work! In no-fly zones, the drone can fly, checked.
>
> Already 2 times successfully installed the super patch, and when flying anywhere the same problem arises. Sometimes, for no [...]

I’m not quite sure what to tell ya. There’s lots of people that have done this on P4P and get Galileo signal on their P4P. Galileo works fine with P4P hardware. Have you had any crashes in the past?

If your bird isn’t safe with the Super-Patcher flight controller I would stop using it but I don’t know why its working that way with your bird. Sorry

sunravers, 2019-08-14:
I get it. There have never been such problems. Did you test waypoints mode after installing the patch?

sunravers, 2019-08-14:
[image]

Information about the drone in the DJI GO 4 application after installing the patch.

brett8883, 2019-08-14:
Everything looks good from that picture. I don’t have a P4P but if this was a problem on P4P we would have heard about here or in FC_Patcher

sunravers, 2019-08-15:
Good. I will calibrate the various sensors several more times and conduct test flights. Why is the firmware version not showing correctly?

brett8883, 2019-08-16:
It helps so that DJI Go doesn't nag you about updating the firmware. However, P4P is on the latest firmware already so its more for the other aircraft at this point

sunravers, 2019-08-18:
Again, during the flight, I started having problems with GPS. Still, I tend to think that the problem is Galileo satellites. I will ask you to release a patch for Phantom 4 Pro v1 without Galileo satellites. I want to test and thus help myself and you.

sunravers, 2019-08-18:
Galileo in Wiki https://en.wikipedia.org/wiki/Galileo_(satellite_navigation)
From 11 July till 18 July 2019 the whole constellation experienced an "unexplained" signal outage with all active satellites showing "NOT USABLE" status on Galileo status page. The cause of the incident originated by an equipment malfunction in the Galileo ground infrastructure, which affected the calculation of time and orbit predictions.

brett8883, 2019-08-19:
> Galileo in Wiki https://en.wikipedia.org/wiki/Galileo_(satellite_navigation)
> From 11 July till 18 July 2019 the whole constellation experienced an "unexplained" signal outage with all active [...]

I think this is an argument against your point though because when that happened Galileo dropped out but it caused no problems because GPS and GLONSS still connected. 

The method to remove Galileo from the patch is known. https://github.com/o-gs/DJI_FC_Patcher/issues/4#issuecomment-417869949

sunravers, 2019-08-28:
Thanks for the link to DJI_FC_Patcher. Using it and Python, I put together my patch, which turned off the processing of Galileo satellites. Now my Phantom 4 Pro flies without problems and has no limits)

Preeminentz, 2019-08-31:
I have the same issue. 
Competent with drones, and a few hundred hours, over 4 years piloting the P4P. 

Have taken it for test flights as my RTH feature is NOT working. 
When i test RTH feature it switches it to atti mode and begins to do it's own thing. also says "Weak GPS signal" some times it does NOT even function to the point of it giving the warning "Weak GPS signal" but as the author has stated it's "Scary to fly" as i feel not in control of her. Never have had this experience and i've used old methods to rid of NFZ and Height restrictions... For some reason i thought id give this ago, i shouldn't of. "Why fix it if it ain't broken" 
Any way's this is a real issue and it's two (2) known cases. P4P uses beware to do short ditance test runs to test RTH features before setting out as per usual on our missions. 

Thanks for all the time/energy my peoples, I'll perhaps revert to the old method till we have a fix.

brett8883, 2019-09-02:
> I have the same issue.
> Competent with drones, and a few hundred hours, over 4 years piloting the P4P.
> 
> Have taken it for test flights as my RTH feature is NOT working.
> When i test RTH feature it switches it to atti mode and begins to do it's own thing. also says "Weak GPS signal" some times it does NOT even function to the point of it giving the warning "Weak GPS signal" but as the author has stated it's "Scary to fly" as i feel not in control of her. Never have had this experience and i've used old methods to rid of NFZ and Height restrictions... For some reason i thought id give this ago, i shouldn't of. "Why fix it if it ain't broken"
> Any way's this is a real issue and it's two (2) known cases. P4P uses beware to do short distance test runs to test RTH features before setting out as per usual on our missions.
> 
> Thanks for all the time/energy my peoples, I'll perhaps revert to the old method till we have a fix.

Agreed. Seems some people have this issue with P4P and P4A. 

Would like to get more data on this issue because I know some people aren’t having the issue at all and others have it sometimes but not others. 

What country or general geo location are you in? Perhaps GNSS settings need to be adjusted for certain regions. @Matioupi is the expert on GNSS stuff. I’ll try to get ahold of him to see what he thinks.

plblacklock, 2019-09-15:
I’m having the exact same issue with my p4p V. 1. The OP couldn’t have described it better. Dropping from gps mode to atti mode in flight despite showing 20+ satellites connected. Also causing compass errors in flight. I suspect the compass issue may be related to the gps problem but it’s only a gut feeling. Everything works fine after switching back to stock firmware. Hope things get ironed out soon.

plblacklock, 2019-09-16:
Forgot to mention, the issue may have started after ascending through the old altitude limit of 1640. Not sure it’s relevant and I’m not going to try again to test but I thought it was worth mentioning. You had also asked another poster about his country and geo location without an answer. I’m in the US, San Jose, California and usually fly within the inner yellow geozone around Reid hillview airport. If that helps....

Preeminentz, 2019-09-16:
> Agreed. Seems some people have this issue with P4P and P4A.
>
> Would like to get more data on this issue because I know some people aren’t having the issue at all and others have it sometimes but not [...]

Hi gang.

Location: Queensland, Australia. 
Since last post i have reverted back to old AC firmware and edited NFZ/Height as i currently had before updating to this method. No issues flies like a dream as always.

Perhaps it may be the location. 

Thanks again. Happy Flying.

plblacklock, 2019-09-16:
So here’s the thing, I’m an absolute noob. Not to piloting but to anything we are covering in this subject. Zero coding skill whatsoever. As for removing the Galileo satellite function as a fix or editing the nfz/height in the stock ac firmware, does anyone have an almost idiot proof guide to follow?

brett8883, 2019-09-21:
> Thanks for the link to DJI_FC_Patcher. Using it and Python, I put together my patch, which turned off the processing of Galileo satellites. Now my Phantom 4 Pro flies without problems and has no limits)

If can you share your patch since you’ve tested it and working? You can even just put in a pull request to add it to Super-Firmware cache if you don’t want to host it. 

I could do this but if you already have a patch you have tested and verified working seems like a better solution.

brett8883, 2019-09-21:
> So here’s the thing, I’m an absolute noob. Not to piloting but to anything we are covering in this subject. Zero coding skill whatsoever. As for removing the Galileo satellite function as a fix or [...]

No there’s a bit of a learning curve anyway you look at it but either I will make one or @sunravers will post the one that has already been tested.

I wish we could figure out why a small number of people have this issue when the vast majority don’t however. Possibly different suppliers for the GPS units? Idk

plblacklock, 2019-09-21:
Thanks for the reply as well as all the help! I suspect you are correct regarding multiple suppliers of the gps module. Funny you mention that as I was just looking at available replacement modules.

plblacklock, 2019-09-24:
Catching up on years of apathy. Downloaded DDD and flashed the p4p with the earliest available firmware. Also downloaded the ancient assistant 2 so now I can theoretically play with debug mode (but I’ll probably steer clear). After flashing with the p4p with 01.01.0203, I reflashed with the latest using the current assistant 2. Only seeing satellite count fluctuate between 13 and 15. From a layman’s point of view, I wonder if this overwrote the Galileo bug issue. Will require extensive flight tests to be sure. Still hoping a fix is coming for super Patcher soon. I owe you guys endless gratitude for giving us all back 100 percent ownership of our birds. Can’t wait to use it again soon!

plblacklock, 2019-09-25:
Eh, spoke too soon. Problem remains even after all of the firmware play. Bug is dug in good and deep. Begging for help.

sunravers, 2019-09-25:
> Eh, spoke too soon. Problem remains even after all of the firmware play. Bug is dug in good and deep. Begging for help.

Do not worry. I compiled a new patch and sent brett8883 for verification. If he approves, then you can install it through a super patcher. Just do not forget to reinstall the firmware V01.05.0600 before installing the patch 2 times. I hope my patch will help everyone who had problems with GPS.

plblacklock, 2019-09-26:
That’s incredible news! Thanks to all for the hard work! Looking forward to the new patch!

brett8883, 2019-09-26:
> Do not worry. I compiled a new patch and sent brett8883 for verification. If he approves, then you can install it through a super patcher. Just do not forget to reinstall the firmware V01.05.0600 [...]

Where did you send it I don’t see anything?

sunravers, 2019-09-26:
> Where did you send it I don’t see anything?

I don’t really understand the work of this site, so I created a branch and uploaded the patch where I could: D
You can download it and add it to your firmwares. A description of the changes in the firmware is attached.
https://github.com/sunravers/DJI_Super-Patcher/tree/sunravers-patch-1

brett8883, 2019-09-26:
> I don’t really understand the work of this site, so I created a branch and uploaded the patch where I could: D
> You can download it and add it to your firmwares. A description of the changes in the [...]

Ok I see it now. Yea it’s even more confusing because of the way I have it set up so no worries. I will integrate into Super-Patcher shortly. I don’t see the description of the changes yet though? Main think I would need to know is if you changed the restart motors while inverted parameters? Would need to tell people that so if not somebody doesn’t try to do it and then crash.

brett8883, 2019-09-26:
You what for now here is the direct link to @sunravers flight controller

[P4P NO GAL RECEPTION FC #03.02.44.08](https://github.com/sunravers/DJI_Super-Patcher/raw/sunravers-patch-1/P4P_03.02.44.08_dji_system.bin)

When loading the firmware to DUMLdore in Super-Patcher instead of choosing the file that Super-Patcher tells you to load, load this one instead.

When checking the flight controller # with NLDapp the correct flight controller # will be 03.02.44.08 if the patch takes. 

This patch only works for Phantom 4 Pro. It WILL NOT work with any other aircraft. Do not try it will any other aircraft not even P4, or P4 advanced. Doing so could result in bad things and best case dumldore wont flash it.

plblacklock, 2019-09-27:
Ok, so when reaching the point where I load the modified bin file to dumldore, the file isn’t visible within the browser window of the location (desktop) in which I saved it. Did a copy paste of the file name into the bottom of the browser window and then hit flash firmware in dumldore. Things just hang in the transferring to device stage. Guessing dumldore can’t find the new bin File because I didn’t save it correctly? Guessing it’s my near windows illiteracy causing problems. Any idea what I might be doing wrong?

sunravers, 2019-09-27:
> Ok, so when reaching the point where I load the modified bin file to dumldore, the file isn’t visible within the browser window of the location (desktop) in which I saved it. Did a copy paste of the [...]

In order for DUMLdore to see the file with the firmware, it is necessary that the file name contains at the end the name _dji_system.bin.
Therefore, do not change the file name when saving to a computer.

sunravers, 2019-09-27:
I made minimal changes in the patch compared to standard firmware. All changes can be seen in the description by reference.
https://github.com/sunravers/DJI_Super-Patcher/pull/1

brett8883, 2019-09-29:
> I made minimal changes in the patch compared to standard firmware. All changes can be seen in the description by reference.
> [sunravers#1](https://github.com/sunravers/DJI_Super-Patcher/pull/1)

Oh ok so you didn't disable the height limit on yours.

brett8883, 2019-09-29:
OK well re-do on this. I have made a patch that specifies using the factory default GNSS settings so it will over write any changes to the GNSS receiver that were made by past patches. 

It has all the same changes made as if you said "no" to both Super-Patcher 2.0 questions. So no Smart battery mod (can be added with parameters in Assistant 2 1.1.2, however) and no stealth and no galileo but everything else is the same. 

I don't have p4p so no way for me to test this. https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/P4P/P4P_2.0_No_gal_standard/P4PSP_2.0_No_gal_standard_03.02.44.99_dji_system.bin

plblacklock, 2019-10-02:
I’m sorry, I should have followed up sooner but I’ve been absolutely swamped with work. I was able to implement the fix and things appear to be operating as hoped. I have yet to see any failures and am slowly regaining confidence with each flight. Many thanks!

sunravers, 2019-10-06:
> I’m sorry, I should have followed up sooner but I’ve been absolutely swamped with work. I was able to implement the original fix and things appear to be operating as hoped. I have yet to see any failures and am slowly regaining confidence with each flight. Will give the latest fix a test run and report back when the schedule loosens. Many thanks!

What version of firmware did you install?

plblacklock, 2019-10-06:
Flashed with latest 01.05.0600 and then used Super Patcher to flash with the modified .bin file you sent to Brett8883. I’ve slowly worked back to my pre Super Patcher flight distances with zero issues. I know you said you did place an altitude limit in the patch but I haven’t tested that. Probably won’t unless I can get out into some serious country. Currently sitting under class B and I can respect that. Again, many thanks for the fix!

sunravers, 2019-10-07:
> Flashed with latest 01.05.0600 and then used Super Patcher to flash with the modified .bin file you sent to Brett8883. I’ve slowly worked back to my pre Super Patcher flight distances with zero [...]

I get it. Confidence in the reliability of the drone will return. I went through all these sensations) It's good that my help came in handy.

brett8883, 2019-10-17:
Just an update on this, in the next version of Super-Patcher Galileo satellite reception will be optional for P4P just like the Smart Battery mod and the Stealth Mod.

Might take a little bit of time to get all the flight controllers made and set up but I will be implemented here soon

Demion, 2019-11-11:
I also have no GPS reception (Atti) on Mavic Pro V01.04.0300. 
Can GPS option be optional and disabled also for Mavic Pro? Thanks.

brett8883, 2019-11-15:
> I also have no GPS reception (Atti) on Mavic Pro V01.04.0300.
> Can GPS option be optional and disabled also for Mavic Pro? Thanks.
> 
> Update:
> Found flight controllers in super-Firmware_Cache.
> Probably will be in upcoming 2.1 version. Thanks for your work.
> https://github.com/brett8883/super-Firmware_Cache/tree/2.0/MavicPro/2.1
> 
> Update2:
> Seems to work better with NoGal.
> But I think `Mavic_SP_2.0_Stealth_NMR_03.02.44.43_YesGal` and `Mavic_SP_2.0_Stealth_NMR_03.02.44.87_NoGal` has some weird parameter changes.
> Others seems to be fine.

You are very observant. Yes Galileo will be optional in 2.1. There will also be some other optional features and new mods in 2.1. You have chosen one with the “NMR” designation which is what I am using to designate another new optional feature. Look for the version that does not have the “NMR” designation. To get the feature set that coincides with the current advertised feature list in 2.0.2 but with NoGal for the no Galileo version.

2.1 is a work in progress that has a few things to work out before release. It’s almost ready, however.

sunravers, 2019-11-15:
How quickly I determined in the summer that the whole problem with GPS was Galileo's signal processing was incorrect. If I were not a programmer and did not create my patch, then some P4P owners would continue to suffer, not knowing the reason why the drone itself goes into ATTI mode.

mtrack8, 2021-04-01:
> How quickly I determined in the summer that the whole problem with GPS was Galileo's signal processing was incorrect. If I were not a programmer and did not create my patch, then some P4P owners would continue to suffer, not knowing the reason why the drone itself goes into ATTI mode.

Thank you very much！

sunravers, 2021-04-01:
> Thank you very much！

🙂

orangpelupa, 2022-09-14:
> How quickly I determined in the summer that the whole problem with GPS was Galileo's signal processing was incorrect. If I were not a programmer and did not create my patch, then some P4P owners would continue to suffer, not knowing the reason why the drone itself goes into ATTI mode.

so what was the cause and what was the fix?

gold-sunshine, 2023-08-24:
my P4P also got intermittent gps signal loss after using super patcher. Using P4P_SP_2.0_03.02.44.32_dji_system.bin。