brett8883/DJI_Super-Patcher #18: Phantom 4 PROv2 NFZ do not work

stpzuev, 2019-08-16:
I install Super Patcher 1.7.5 to Phantom 4 PROv2 to avoid NFZ. But in does not work. When I try to fly in NFZ drone automatically lands in 20 seconds.
I made all steps from instruction. Firmware V01.00.1500. FC after patch is 03.03.04.36

May be i need to change some parameters in DJI Assistant?

brett8883, 2019-08-17:
> I install Super Patcher 1.7.5 to Phantom 4 PROv2 to avoid NFZ. But in does not work. When I try to fly in NFZ drone automatically lands in 20 seconds.
> I made all steps from instruction. Firmware V01.00.1500. FC after patch is [03.03.04.36](tel:03.03.04.36)
> 
> May be i need to change some parameters in DJI Assistant?

I’m aware of this issue and the solution that seems to work is by using a modified app that has the NFZ database removed. Looks like the P4PV2 is still able to get NFZ info from the app and is different from other aircraft in that regard. Looking into how that process works and how to stop it. 

You’ll notice that if you remove the app and try to take off that the aircraft NFZs have been unlocked so it has something to do with the app. 

I don’t have a P4Pv2 so I have to rely on other people to test out the P4Pv2 hacks so I can use all the help I can get as far as figuring out if there’s a way to stop the aircraft from getting NFZ updates from the app. 

But thanks for bringing it up and so other people who see this will know to use a modified app until we can figure out a way so that it doesn’t get NFZ info from the app

stpzuev, 2019-08-18:
Yes, it looks like you said. P4PV2 gets info about NFZ from app. I try to fly in my apartments, they are inside NFZ,  aircraft doesn't see GPS but app is able to locate themselves and start to autolanding aircraft.

Do you have any idea how to remove NFZ from app?

brett8883, 2019-08-19:
> Yes, it looks like you said. P4PV2 gets info about NFZ from app. I try to fly in my apartments, they are inside NFZ, aircraft doesn't see GPS but app is able to locate themselves and start to autolanding aircraft.
> 
> Do you have any idea how to remove NFZ from app?

https://github.com/Bin4ry/deejayeye-modder

brett8883, 2019-10-20:
P4PV2 files have been updated to hopefully solve this issue.

Wangk1ng, 2019-10-25:
I have helped my companion to solve many problems in the no fly zone, which I have also encountered. The problem is at this stage.
**adb shell mount -o remount,rw /vendor
adb shell mkdir /vendor/bin
adb push dummy_verify.sh /vendor/bin/**
`echo 1 file pushed. **0.0MB/s**，`
no other misstatements，until the flash is successful， NFZ is not work.
Then I switched to another computer.Flash again,
`echo 1 file pushed. **0.3MB/s**，`
NFZ is OK.
I can use any version of djigo4 app

brett8883, 2019-10-29:
> I have helped my companion to solve many problems in the no fly zone, which I have also encountered. The problem is at this stage.
> **adb shell mount -o remount,rw /vendor adb shell mkdir /vendor/bin [...]

Sweet thanks for the update. Not sure what happened there but sounds like Super-Patcher just wasn’t successful using that PC. 

Glad you got it to work. 

Looks like this issue has been resolved if anybody else has issues please reopen or make a new issue. I’m gonna close this one

closed 2019-10-29 by brett8883