brett8883/DJI_Super-Patcher #20: Missing bin. file during Super-Patcher 1.9 installation

BossTon62, 2019-08-22:
There is no file "MavicPro_MPP_super_patcher_03.02.44.10_dji_system.bin" in directory during installation DJI Super-Patcher 1.9
All installing was made by instruction step by step. 
Mavic 1. Win 7

Weirdoz, 2019-08-22:
Use a Win10/64 machine, win7 does not work proper with superpatcher.

BossTon62, 2019-08-22:
Is i understood correctly that problem of lack of bin. file is window problem?
Jackson mod work perfect.
Could download file from some place by myself?
Because i downloaded it in your directory Super-Firmware_Cache/MavicPro_Super_Patcher_FC/
and used it and Installation was successful, but I'm afraid it was a wrong file.

brett8883, 2019-08-22:
> There is no file "MavicPro_MPP_super_patcher_03.02.44.10_dji_system.bin" in directory during installation DJI Super-Patcher 1.9
> All installing was made by instruction step by step.
> Mavic 1. Win 7

I just tried it on Windows 10 and the correct file did generate to the main Super-Patcher folder. Maybe you just didn't see it in that directory? I wouldn't understand why this particular issue would happen on Windows 7 and not windows 10. This just uses busybox 

At any rate you figured it out and got the correct file. Super-firmware_Cache is a bit unorganized I will admit and I will organize it time permitting but it does take time since I don't want to break the links to the those files. 

Just to reiterate the correct file for Mavic Pro is can be downloaded manually from https://github.com/brett8883/Super-Firmware_Cache/raw/master/MavicPro_MPP_super_patcher_03.02.44.10_dji_system.bin

You don't have to worry, you have the correct file. I take great caution not to make more than one file with the same flight controller # for each aircraft so there isn't this concern. The files always have the flight controller # in the name. 

Nice job figuring out how to solve the issue yourself!