brett8883/DJI_Super-Patcher #21: wget error

biosdi, 2019-08-25:
During auto_install.bat execution I receive this error:

wget: can't execute ssl_client

What does it mean and how can I solve it?

Thanks

(I'm on Mac running WIN10 on Parallels)

brett8883, 2019-08-25:
Is this before or after you choose your aircraft type?

biosdi, 2019-08-25:
Soon after launching auto_install.bat :(

brett8883, 2019-08-25:
So it’s busybox that doesn’t work in the VM. You might have other issues using a vm with Super-Patcher given how the USB ports work in VMs so the best solution is to just run bootcamp on the Mac and install Windows natively. [How to enable bootcamp to run Windows on a Mac ](https://support.apple.com/en-us/HT201468)

You can try using the offline version of Super-Patcher. It doesn’t have the same updates as the 1.9 version but it ultimately gets you to the same place if you can get it to work.

Again this isn’t as good as running bootcamp or simply using a real PC (as in not virtual 😉) but it’s faster and won’t take very long to figure out if it’s gonna work or not.

biosdi, 2019-08-25:
Thanks! 

Where can I find the offline version?

And, will I miss any mandatory update for Mavic using offline version?

Thanks

brett8883, 2019-08-26:
Sorry I thought I meant to include the link [Super-Patcher 1.7 offline](https://github.com/brett8883/DJI_Super-Patcher/tree/1.7_OFFLINE)

This is a self-contained version. It doesnt have the error handling or logging that the master version has but the flight controller are the same so you'll end up getting the same thing if its successful.

Weirdoz, 2019-08-26:
I use VMWare Fusion on a MBP and with win10/64 all worked well, like on a real win machine.