brett8883/DJI_Super-Patcher #25: NLD App is missing in the Tools Folder

Weirdoz, 2019-08-26:
In the 2.0, the NLD app isn‘t downloaded in the Tools Folder on Win10. But you can easyli add it by yourselfe.

brett8883, 2019-08-26:
It gets downloaded at runtime directly from NLD to that folder. Is it not opening for you when you run it? Just tried it and it works. Tried earlier on somebody else's computer and it works. Try again

Weirdoz, 2019-08-26:
Nope, i tried it 3 times, and the nld is always missing in the tools folder, but if add it manual it runs.

brett8883, 2019-08-26:
This is in your appdata folder?

Weirdoz, 2019-08-26:
[image]
no it‘s not there...

suryo88, 2019-08-26:
just install SP 2.0 on MPP. its easy and flawless
thanks

Weirdoz, 2019-08-26:
> just install SP 2.0 on MPP. its easy and flawless
> thanks

What you wanna tell me? I know it is easy, but nld is missing in the tools folder. I know i can download it by my own, i already had alicense for my mavic.

Weirdoz, 2019-08-26:
Nope, you can see it in my snapshot it isn‘t there. I‘ve tried it 3x times.

suryo88, 2019-08-26:
NLD is in there, you dont have to download anything else.

Weirdoz, 2019-08-27:
> I’m not sure what’s happening @Weirdoz. It works for me and seems like it works for other people too. Do you have NLD blocked by a firewall maybe? At least you were able to fix it for yourself.

Nope, it is not blocked, otherwise i could not download it by my own. 
Like i said it is a fesh win10, just 1 day old.

Weirdoz, 2019-08-30:
Thats my Log, Problem still exists.

START SUPER_PATCHER 
28.08.2019_19:12:59.53 
I have verified internet access to GitHub 
19:13:03.41 
 
Download NLD app 
--2019-08-28 19:15:14--  https://nolimitdronez.com/downloads/nldapp.zip
Resolving nolimitdronez.com (nolimitdronez.com)... 85.184.161.145
Connecting to nolimitdronez.com (nolimitdronez.com)|85.184.161.145|:443... connected.
ERROR: cannot verify nolimitdronez.com's certificate, issued by 'CN=Sectigo RSA Domain Validation Secure Server CA,O=Sectigo Limited,L=Salford,ST=Greater Manchester,C=GB':
  Unable to locally verify the issuer's authority.
To connect to nolimitdronez.com insecurely, use `--no-check-certificate'.

Weirdoz, 2019-08-30:
I have tried it again today, and all worked perfect, NLD app was downloaded and also the patching of the ac without any probs. 
Thx brett for your awesome work, i have already sent some bucks to your PP account.