brett8883/DJI_Super-Patcher #28: Here's a proposal

hjchjcjh, 2019-08-27:
I want DJI_Super-Patcher to support third-party batteries and display voltages in apps， because smart batteries are too expensive

Weirdoz, 2019-08-27:
> This is a feature request.
> Background: Third-party batteries can greatly improve the battery life without affecting the performance of the maivcpro, and the price is low, and the modification is of great value.
> I want to solve the problem: when using a third-party battery, the battery voltage cannot be displayed on the djigo4, and the remaining battery power is not visible.
> Test results: 3S2P 21700 battery (weight 400g) 58min.
> Many geeks in China use mavicpro third-party batteries to challenge 10,000 meters in height.
> That's very nice of you
> [image]

Do you have more specific facts about your mod, or a website about it? Thx

hjchjcjh, 2019-08-27:
This is a feature request.
       Background: Third-party batteries can greatly improve the battery life without affecting the performance of the maivcpro, and the price is low, and the modification is of great value.
       I want to solve the problem: when using a third-party battery, the battery voltage cannot be displayed on the djigo4, and the remaining battery power is not visible.
      Test results: 3S2P 21700 battery (weight 400g)   58min.
       Many geeks in China use mavicpro third-party batteries to challenge 10,000 meters in height. 
     That's very nice of you 
[image]

hjchjcjh, 2019-08-27:
Refer to this video: https://www.youtube.com/watch?v=HDt_UoSlGdI
       The main problem is the use of third-party batteries. DJigo4 has no voltage display, but it can read the voltage in the DJI assistant. We need to output the voltage parameters in the DJI assistant to the power display bar of DJigo4. This involves the modification of firmware communication protocol. I don't understand it very well. Or do you have any good suggestions? Thank you.

hjchjcjh, 2019-08-27:
I am a junior reverse engineer

hjchjcjh, 2019-08-29:
If I change it to  non-DJI battery, the voltage will only be displayed in the DJI Assistant's parameters and will not be displayed in the DJIGO4 main interface or remote control. I can't see the remaining power when I fly.

In order to allow the non-DJI battery to see the remaining power during the flight, I hope to display the voltage on the DJIGO4 or remote control during the flight.

brett8883, 2019-09-15:
Hmm... won’t be right away but will see. Might be an app issue

brett8883, 2019-10-03:
What if you welded connectors onto a DJI smart battery chip so that you could connect and disconnect battery cells from the DJI Smart Battery chip easily? 

That way you keep the smart battery chip between flights but disconnect the actual cells and swap in new cells. The way Assistant 2 does this is likely dependant on having the aircraft in Assistant mode and it can't fly in that mode. Getting the raw battery data transmitted to the app over occusync might be something requiring the reworking of the entire air encoder system and that's more work than it's probably worth. 

By modifying the hardware to conform to the existing system I think you could get the same benefit possibly without the need to rework the entire system. 

Just a thought...

Demion, 2019-11-08:
I managed to show battery charge percent in RC using 55 2C packet from [mavic_bat](https://github.com/sin5678/mavic_bat ) but still showing N/A in DJI GO.
So this means RC should receive data over occusync but app detects there is no original battery.

stekiro, 2021-05-11:
hello @brett8883 i have realized your thought 3 years ago and now i see that discussion... i have flash my m1p with super patcher but how to change the battery capacity ?i have set no dji battery in debug mode and i have try to use a EV2300 for connect to battery electronic in l2C and is ok but no read no write... have you a solution for change capacity ? thank you for the super job

zwgtlx, 2021-05-13:
电池容量修改在ev2300可以修改！发自我的iPhone------------------ Original ------------------From: stekiro  ***@***.***>Date: Wed,May 12,2021 0:15 AMTo: brett8883/DJI_Super-Patcher ***@***.***>Cc: Subscribed ***@***.***>Subject: Re: [brett8883/DJI_Super-Patcher] 3rd Party battery mod (#28)
hello @brett8883 i have realized your thought 3 years ago and now i see that discussion... i have flash my m1p with super patcher but how to change the battery capacity ?i have set no dji battery in debug mode and i have try to use a EV2300 for connect to battery electronic in l2C and is ok but no read no write... have you a solution for change capacity ? thank you for the super job

[
{
"target": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838768025",
"url": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838768025",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

zwgtlx, 2021-05-13:
密码是多少，你可以搜索一下，我忘记了！但是就是在ev2300里修改。发自我的iPhone------------------ Original ------------------From: stekiro  ***@***.***>Date: Wed,May 12,2021 0:15 AMTo: brett8883/DJI_Super-Patcher ***@***.***>Cc: Subscribed ***@***.***>Subject: Re: [brett8883/DJI_Super-Patcher] 3rd Party battery mod (#28)
hello @brett8883 i have realized your thought 3 years ago and now i see that discussion... i have flash my m1p with super patcher but how to change the battery capacity ?i have set no dji battery in debug mode and i have try to use a EV2300 for connect to battery electronic in l2C and is ok but no read no write... have you a solution for change capacity ? thank you for the super job

[
{
"target": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838768025",
"url": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838768025",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

zwgtlx, 2021-05-20:
你首先要找到你所用电池的超级密码，否则无法进入超级参数表进行修改电池容量等参数的设置。 记得这是德州的芯片，你也可以去相应的论坛，去询问一些高手，他们通常有德州电子芯片的电池保护芯片的密码。发自我的iPhone------------------ Original ------------------From: norfeas  ***@***.***>Date: Wed,May 12,2021 1:25 AMTo: brett8883/DJI_Super-Patcher ***@***.***>Cc: Subscribed ***@***.***>Subject: Re: [brett8883/DJI_Super-Patcher] 3rd Party battery mod (#28)

hello @brett8883 i have realized your thought 3 years ago and now i see that discussion... i have flash my m1p with super patcher but how to change the battery capacity ?i have set no dji battery in debug mode and i have try to use a EV2300 for connect to battery electronic in l2C and is ok but no read no write... have you a solution for change capacity ? thank you for the super job

Here you can find some information.... o-gs/dji-firmware-tools#17

[
{
"target": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838868553",
"url": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838868553",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

zwgtlx, 2021-05-20:
https://haokan.hao123.com/v?vid=13560618147275273313&pd=haokan_share&context=%7B%22cuid%22%3A%22_uShi_i2Su_gaS8cj8vOtgup2f_C8B8qYP-jajuN-il3uHi50iS_u_8s1OJfiWaTY_DmA%22%7D发自我的iPhone------------------ Original ------------------From: norfeas  ***@***.***>Date: Wed,May 12,2021 1:25 AMTo: brett8883/DJI_Super-Patcher ***@***.***>Cc: Subscribed ***@***.***>Subject: Re: [brett8883/DJI_Super-Patcher] 3rd Party battery mod (#28)

hello @brett8883 i have realized your thought 3 years ago and now i see that discussion... i have flash my m1p with super patcher but how to change the battery capacity ?i have set no dji battery in debug mode and i have try to use a EV2300 for connect to battery electronic in l2C and is ok but no read no write... have you a solution for change capacity ? thank you for the super job

Here you can find some information.... o-gs/dji-firmware-tools#17

[
{
"target": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838868553",
"url": "https://github.com/brett8883/DJI_Super-Patcher/issues/28#issuecomment-838868553",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]