brett8883/DJI_Super-Patcher #30: Where's the file called “auto_install.bat”

aldo962, 2019-08-28:
Hello, after unzipping the file and running the exe, i can't find the file called “auto_install.bat”, is it missing or the procedure has changed ?

aldo962, 2019-08-28:
Well no problem, I supposed that Super-PatcherV2.0.exe was only for the initial setup. Anyway I want to thank you for the beautiful work you have done to simplify al these procedures. If I can suggest you a thing for the readme it's better to tell to unzip the package on the C: drive of the PC. I have a lot of HDs on my PC and I'm used to divide software by argument on drives (and drone software for me is on drive D:). Well if you install it on D drive you can't run it correctly, but maybe you are already  aware of it. Thank you again. Aldo

aldo962, 2019-08-28:
Thank to you for your  work !

brett8883, 2019-08-29:
Issue was fixed closing issue

closed 2019-08-29 by brett8883