brett8883/DJI_Super-Patcher #31: Mavic Pro 1 - BattMod - Versions

goguaitor, 2019-08-29:
Hello!

Thank you for your work, i run superpatcher with my mavic pro 1, from 1.03.900 to 1.04.300 previously with assitant.
All runs OK and i see that install BattMod 03.02.44.12 version.
Some of my frinds tell me that your version installed is BattMod 03.02.44.11

Why this difference ?

thank you so much

anonim-o, 2019-08-29:
A mi me descarga la version BattMod 03.02.44.11es la que instale porque pasa esto?
Otra cosa aller me marcaba el tiempo de vuelo que tenia de bateria pero hoy ya no.
Muchas gracias.

brett8883, 2019-08-29:
There’s a different file for every possible combination of options since the tools can’t edit FCs directly. To achieve this I have used the last two numbers to act as the indicators for the options associated with the FC. The first number indicates if the battery mod option is selected. 1 for yes 2 for no. The second and final digit indicates the same for the Stealth mod. .11 indicates battery mod AND stealth are selected. .12 means only battery mod is selected. .22 would bean neither option is selected etc. 

It is the SECOND to last digit that indicates the battery mod. Therefore both .12 and .11 have the battery mod active.

closed 2019-08-29 by goguaitor