brett8883/DJI_Super-Patcher #33: Antivirus False positive?

dissident76, 2019-08-30:
https://www.virustotal.com/gui/file/02f8f14ae67abb4a36fb15ebdf545e4c990bca5c6af74d1549bc6af6fac8e129/detection

brett8883, 2019-08-31:
I’m having the same issue as well. I will have a permanent fix shorty but for now the source files for Super-Patcher 2.0 are on the 2.0 branch https://github.com/brett8883/DJI_Super-Patcher/tree/2.0. 

Download or clone the branch and run the auto_install1.cmd. It’s the same thing as the executable I just thought the exe would be less cluttered. I think exe files set off antivirus software especially when a Linux tool is inside them (busybox). 

For now use the auto_install1.cmd on the 2.0 branch till I can figure out a better solution.

dissident76, 2019-08-31:
Cool, thanks for the explanation

brett8883, 2019-09-02:
Issue marked as resolved because a work around has been provided but the underlying issue is still present, therefore, the issue will remain open.

brett8883, 2019-10-03:
I reverted the exe file back to batch files so I will close this issue as its no longer relevant

closed 2019-10-03 by brett8883