brett8883/DJI_Super-Patcher #34: The system cannot find the path specificed.

Hsh-CODER, 2019-09-04:
After I ran the .exe file and it said connecting to GitHub,the message 'The system cannot the path specificed' showed up,the the application close itself.

racecat007, 2019-09-04:
same problem was here, but was cleaned by starting the tool from the User Desktop

brett8883, 2019-09-04:
Yes the exe file needs to be run from a location in the user path such as the desktop, documents, downloads, pictures etc.

bmasephol, 2019-09-24:
I'm having an issue using both the .exe and the .cmd version.  I'm getting an error on the wget call to download the 2.0.zip file.  If i put the URL manually into a browser it works.  I've picked apart a bit of the script to figure out what its doing and tried to run the commands by hand and I get this error...

PS C:\Users\XXXXXXXX\AppData\Roaming\DJI_Super-Patcher> .\busybox.exe wget https://github.com/brett8883/Super-Tools/archive/2.0.zip
Connecting to github.com (192.30.253.113:443)
ssl_client: write error: Invalid argument
wget: error getting response

Any ideas?

brett8883, 2019-09-27:
> I'm having an issue using both the .exe and the .cmd version. I'm getting an error on the wget call to download the 2.0.zip file. If i put the URL manually into a browser it works. I've picked apart [...]

Not really? I just used PowerShell to do the same thing and it worked for me. 

Since you are reverse engineering it just download super-tools manually unzip it into C:\Users\XXXXXXXX\AppData\Roaming\DJI_Super-Patcher.

Then just comment out the bit to download and unzip it. 

Presto!

[DJI_Super-Patcher-master.zip](https://github.com/brett8883/DJI_Super-Patcher/files/3664433/DJI_Super-Patcher-master.zip)

You'll

brett8883, 2019-10-26:
This should be fixed now. Have hot fixed it in Super-Patcher 2.0 and higher.

closed 2019-10-26 by brett8883