brett8883/DJI_Super-Patcher #35: Mavic Pro 1 - Motors will restart when aircraft is inverted is not working

racecat007, 2019-09-04:
The feature "Motors will restart when aircraft is inverted" is not working.
It is still possible to turn off the motor by turning it upside down and not possible to start the motor in inverted position.
Any idea why not ?
thx

brett8883, 2019-09-04:
Doesn’t sound like the patch has taken. This is a hard-coded value and so it the patch is present it actually ia not possible to turn it off. Please ensure that the flight controller # number in the NLDapp matches the flight controller # indicated by Super-Patcher. 

Before trying a CSC in flight it is always a good idea to test this on the ground by removing the props and arming the motors. Then flip the drone upside down while the motors are still running. If the motors continue to run then you know the patch has taken. You can further verify by using a csc to cut the motors and then restart them while holding it inverted.

racecat007, 2019-09-04:
Found the problem ;-) (but it is not a problem; but the possibility to disable the function)
with parameter: 
 g_config_fdi_open_close_auto_stop_motor_check = 0
it´s possible to disable the function "Motors will restart when aircraft is inverted"
very cool - if needed - set parameter to 1

brett8883, 2019-09-05:
Yea that’s the parameter but it’s not visible with 01.04.0300 so you aren’t on the correct firmware. You’ll need to flash the correct firmware twice and then re-do Super-Patcher.

brett8883, 2019-09-05:
Reopen if you still have questions I don’t ever check the closed issues

racecat007, 2019-09-05:
[image]

i think the parameter which you mean is **g_cfg_debug_motor_no_start_motor_check**
this parameter cannot be changed but for the other parameter**g_config_fdi_open_close_auto_stop_motor_check**
it´s possible :-)

brett8883, 2019-09-06:
glad you got it figured out

closed 2019-09-06 by brett8883