brett8883/DJI_Super-Patcher #37: Fly away with mavic (I have log)

flaviny, 2019-09-07:
Today I had two problems with Mavic, in my first test.

In a first flight, I raised up to 1.8 km. 
After that, I lost the application connection, but rc remained connected.
Mavic returned with an RTH option. The control remained connected and I set the drone down normally, just by RC.
I changed the battery for the second flight.
I started the engines and after they started, I climbed about 30 meters.
After that, the drone began to go left on its own, with some speed, as if it had no GPS. After that he descended at full speed toward the ground. I was very lucky that he collided with a tree and apparently only suffered a few scratches. I have the log of this flight and from what I analyzed, the enternde drone was 517 meters. It was from zero to 517 meters, according to the log.

I would like to share the log so that someone can analyze who has more knowledge than me.

brett8883, 2019-09-08:
Sorry to hear. Post a link to the log here.

flaviny, 2019-09-14:
I can extract the log of others places, such as the drone, If necessary

brett8883, 2019-09-14:
Yea if you could include the dat file from the “fall” flight. The other flight just shows lots of interference and the drone complains that the antennas might not be pointed at the aircraft. Nothing out of the ordinary really and there isn’t much data that gets collected about signal in the first place. Your second flight is another story. Lots of odd things happening but hints at an imu failure of some kind.

flaviny, 2019-09-14:
Please, how I extract the .dat file?

brett8883, 2019-09-17:
I had an expert look at this log file for the fall and the file is corrupted. Hence why I said “lots of weird stuff.” If you still have it on your mobile device I would download it again and upload it directly to this site without renaming it or anything https://www.phantomhelp.com/LogViewer/Upload/ you can then share the URL to the uploaded file. This ensures the data isn’t corrupted in transfer

brett8883, 2019-10-22:
This issue has gone cold as I haven't heard back for the author. I will close this issue because there isn't enough information to add value and figure out what happened. You can reopen this issue or start a new one if you'd like but please repost the flight log if you do

closed 2019-10-22 by brett8883