brett8883/DJI_Super-Patcher #39: P4 起飞后不受控制

cfso1983, 2019-09-08:
我的P4 起飞后不受控制   已经在两台飞机上验证过了

Old-line, 2019-09-09:
[image]

fanxiang83, 2019-10-17:
遥控器不起作用还是因为gps搜索不到而漂移？你的描述太简单了

cfso1983, 2019-10-17:
解锁后一飞冲天

fanxiang83, 2019-10-17:
那你是怎么把飞机收回来的。。

Wangk1ng, 2019-10-25:
I've also had this problem. After taking off, It's lost control and flew out of control. Then I turned over and fell down.
One of the reasons I found the problem was that I didn't wait for enough GPS to appear, and one of the other four cells of the battery had a low voltage. After replacing the battery and waiting for GPS to appear, the flight is normal.
_
I‘m Chinese. The above passage is by translation software.

cfso1983, 2019-10-26:
> 那你是怎么把飞机收回来的。。

我遇到过三次   一次直接撞房子  一次在屋里面解锁直接撞屋顶  手接住了   一次是起飞解锁 自己就往天上跑 油门到底自己熄火坠地

cfso1983, 2019-10-26:
> I've also had this problem. After taking off, It's lost control and flew out of control. Then I turned over and fell down.
> One of the reasons I found the problem was that I didn't wait for enough [...]

你可以加我微信5938743

yuliang521, 2019-11-07:
Yes in Ati mode，but not with GPS，hope to fix it
姿态模式下会一直上升，有GPS信号就不会