brett8883/DJI_Super-Patcher #40: Increased descenting speed for autolanding don't work - be careful guys.

wwarne, 2019-09-21:
_I have "Spark" so can't say about other models_

Among others parameters in DJI Assistant there is **g_config_landing_auto_landing_vel_L2**
It's the speed of auto-landing above 20 meters. By default it's 3 meters/sec.
I've changed it to 6 m/s but it didn't apply. Auto-landing happens only at 3 m/s. 
It can surprise you then you're flying very high and suddenly lose connection.

Test this thing is easy - just fly away and initiate "return to home". Then see at what speed your spark is descending.

brett8883, 2019-09-21:
This doesn’t have anything to do with Super-Patcher and at least with Mavic Pro has to do with having precision landing enabled. Disable precision landing to enable the ability to set different velocities for the auto-landing levels when multiple landing levels is enabled.  

Or just manually land the aircraft when flying high altitudes.

brett8883, 2019-09-21:
This is not an issue with Super-Patcher so I’m classifying it as a question for now

wwarne, 2019-09-23:
@brett8883 Yes, I suppose it's a problem inside of dji's firmware 01.00.0900. 
I found information that this parameter is working on a different firmware (person who said this was using mixed-firmware mod for Spark).

I've tried to summarize my thought about situations in which descending speed is very important so:

1. You are flying at high altitude and your connection is lost
- Drone performs RTH.
- It starts to land at 3 m/s near home point
- Controller reconnects to the drone, you are able to abort landing and land the drone manually.

2. You are flying and have low battery charge  (or it's just a voltage drop if you're pushing your motors at 100% ). Voltage drop can occur even on 30 or 50% or more.
- Drone performs emergency low battery landing
- You can't abort this type of landing.  So you could have land at 6m/s safely but it might be not enough energy to land at 3 m/s. All you can do is move drone horizontally and pray.
- **I think Smart Battery Mod will help with this scenario.** 

btw Super-Patcher is totally amazing. So easy to use and It's like I've got a different version of my drone - the better one. Thank you a lot!!

wwarne, 2019-09-25:
> Did you disable precision landing to see if your 6 m/s parameter applied? That was what did it on Mavic Pro. When you have precision landing enabled even if you didn’t take off using precision take off the auto-landing mode is changed it what the system called “advanced landing” mode instead of “auto-landing” which is what the parameter you’ve Identified is for.

I didn't find anything about precision landing in my dji go app. (Probably it's because DJI doesn't declare Spark to have it). But I did disable obstacle avoidance system.
Probably I need to try my little experiment again but in Sport Mode. Maybe there will be difference. I'll post here with the new results.

wwarne, 2019-09-26:
I took some tests and results are:
- Spark in GPS mode, obstacle avoidance system is on - Automatic landing at 3 m/s
- Spark in GPS mode, obstacle avoidance system is off - Automatic landing at 3 m/s
- Spark in Sport mode, obstacle avoidance system is off - Automatic landing at 3 m/s

brett8883, 2019-09-29:
> I didn't find anything about precision landing in my dji go app. (Probably it's because DJI doesn't declare Spark to have it). But I did disable obstacle avoidance system.
> Probably I need to try my [...]

Interesting. It's not in the OA sensors menu of the settings? 

I don't have a Spark but its enabled by default in the stock firmware

	
		"index" : 476,
		"typeID" : 0,
		"size" : 1,
		"attribute" : 36,
		"minValue" : 0,
		"maxValue" : 1,
		"defaultValue" : 1,
		"name" : "g_config.landing.adv_landing_enable"

Try turning it off using parameters if its not in the Go 4 app. 

      g_config_landing_adv_landing_enable=0

wwarne, 2019-09-30:
@brett8883 Thank you, I will try it as soon as the weather outside improves.

wwarne, 2019-10-05:
Unfortunately this setting doesn't seem to have any effect. 
Auto-landing max speed is 3 meter/sec no matter what. 
I tried sport mode and normal mode. Obstacle avoidance was disabled in dji go. 
As far as I can tell after watching some YouTube videos about testing Spark's landing - it has some sort of precision landing but you can't change its behavior via dji go app (we just don't have this menu settings). Probably firmware 01.00.0900 doesn't allow to change it at all.

brett8883, 2019-10-08:
Right you have to turn it off from Assistant 2 1.1.2 in debug mode if it’s not in DJI Go 4. At any rate when the drone returns to the home point you should be able to regain radio connection since it will be right above you and then you can cancel RTH and land manually

BorisPlintovic, 2019-10-15:
Spark parameter **g_config_landing_auto_landing_vel_L2** does not work on firmware version 0900, only on 0300.
Parameter **g_config_landing_auto_landing_vel_L1** work fine

Likewise, parameter **g_config_go_home_gohome_idle_vel** will not work if the sensors are on. You must also use **g_config_go_home_avoid_enable=0** to automatically turn off sensors in RTH

brett8883, 2019-10-16:
> Spark parameter **g_config_landing_auto_landing_vel_L2** does not work on firmware version 0900, only on 0300.
> Parameter **g_config_landing_auto_landing_vel_L1** work fine
>
> Likewise, parameter [...]

Thanks Boris!