brett8883/DJI_Super-Patcher #43: Battery and superparcher

majentico, 2019-10-02:
Hi , first of all old like to congratulate you for you project, it’s and amazing idea.
I installed superparcher in a mavic pro platinum few hours ago. So I don’t had time to use it properly.
But I have a battery that is not updated, 
My question is what I have to do with it. 
I also would like to know if the battery has older firmware than the 10.04.300 what’s steps are necessary . And also for battery’s whit a newer firmware than the 10.04.300 
Thanks for your help and for your nice project and for the time you spend helping people.
Sorry for my English.

brett8883, 2019-10-02:
Thanks! 

The Super-Patcher Smart Battery mod is a modification to the way the aircraft reacts to inputs from the smart battery. It is not a modification to the smart battery itself. You do not need to update the batteries to use them with Super-Patcher and still get the benefit of the Smart battery mod. 

If you would like to update all your batteries so they are on the battery firmware supplied in 01.04.0300 you can flash the stock firmware for 01.04.0300 again for each battery. This will undo Super-Patcher but once you have all the batteries updated to 01.04.0300 you can redo the Super-Patcher process to install Super-Patcher once again. 

This is not required. If you do not update the batteries everything will work normally. The instructions above are only if you insist on having all the batteries on the same firmware version as your aircraft. 

Super-Patcher does not touch the firmware on the batteries. 

Let me know if you have any other questions

majentico, 2019-10-02:
Thanks you very much for your response. I will like to know a few things more.
The geo zones are disabled just with the súper parcher or is needed to touch anything in the asistant 
I tried today to flay with the new parcher and I had lot of desconections of the mobile phone (iPhone) the phone was new , can be its fault , and also used the ogt cable, I notice that when I was in air plane mode all was ok ( but dint had too much battery to test more time that)

brett8883, 2019-10-02:
> Thanks you very much for your response. I will like to know a few things more.
> The geo zones are disabled just with the súper parcher or is needed to touch anything in the asistant
> I tried today to flay with the new parcher and I had lot of desconections of the mobile phone (iPhone) the phone was new , can be its fault , and also used the ogt cable, I notice that when I was in air plane mode all was ok ( but dint had too much battery to test more time that)
> It’s any way to know if I’m really in the fcc mode ?
> It’s any way mi drone don’t send information to DJI.
> Really thanks for your help and for that magnificente job you are doing.

Thanks!

GeoZones are automatically disabled with Super-Patcher. Nothing else to do.

I find using the full sized USB OTG on the bottom of the remote works better than the cable they supply with the drone. Be sure the connector is firmly attached in the iPhone. 

There are supposedly some methods to figure out if in FCC mode but I live in USA so I have never seen CE mode. I recommend setting the FCC + Boost setting and it should be obvious that it is working. 

The data transfering from the app to DJI is an app issue and outside my area of expertise so I will defer that question to someone else but I use an app called LockDown for iOS that allows me to set up a firewall on my iOS device to prevent the app from connecting to DJI APIs.  If this is enough I cannot say but it makes me feel more comfortable.

Cheers

majentico, 2019-10-02:
Thanks you a lot 
Cheers