brett8883/DJI_Super-Patcher #44: Inspire 2 losing satellite

calstrin, 2019-10-14:
Hey Brett.
I have installed your super patcher for the Inspire 2 and am having a constant issues losing GPS.  Any suggestions or do you have an update for this to fix it?

brett8883, 2019-10-15:
Sorry about the issue. I am working on making Galileo an optional mod. Seems to be an issue for a very small group of people but this is the first I’ve heard of an issue with Inspire 2. While I am working on rolling out this option for everyone I can send you a flight controller without Galileo. Did you want Smart Battery mod? Did you want stealth? I have all combinations made already so just let me know which one you want.

wanghuakangjq, 2019-10-15:
Inspire 2

> 嘿布雷特。
> 我已经为Inspire 2安装了超级修补程序，但是丢失GPS时经常遇到问题。有什么建议吗，或者您有更新来解决此问题？

Hi！calstrin 
My Inspire 2 is now experiencing the problem of ADB failure。when I use ADB for a few seconds, the plane starts to chirp and the LED lights turn red. I would like to ask you how to solve this problem. Thank you very much.

brett8883, 2019-10-16:
@wanghuakangjq if you are having a separate issue please start a new issue. Having the issues organized under the correct title makes it easier for users that might be having the same issue to get the information they need.

calstrin, 2019-10-16:
Thanks Brett,
I was down in the Southern Hemisphere when it was really acting up.  I flew it today and it was pretty rock solid, didn't disconnect except for a couple seconds then went back to normal. 
If you could send one that has that option to turn off I'd be very grateful.  Yes on the smart battery, (is that so it doesn't fly home on it's own?) . Sorry I forget the explanations.    and no on the stealth.
Thanks

brett8883, 2019-10-17:
The Smart Battery mod disables the low battery and critical low battery warning levels by default. These settings can be adjusted to however you want it. It completely disables low battery forced auto-landing (except at the emergency voltage level which is 3.0v per cell it will auto-land but this can be disabled as well if you want). 

If Smart RTH is enabled in DJI Go 4 it will auto-RTH but only just in time before the battery dies. This is not forced and can be ignored in DJI Go 4. You can also turn of Smart RTH and it won’t auto-RTH in the first place. 

I’ll send you that flight controller shortly.

brett8883, 2019-10-20:
Ok here's the Inspire 2 with smart battery mod and Galileo disabled 

https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/I2/I2_SP_2.0/2.0/NoGal/I2_Batt-Mod_NoGal_03.03.99.99_dji_system.bin

The flight controller# will be 03.03.99.99 when you check with NLDapp. You'll need to do the Super-Patcher process again (1 in main menu) and then just flash this file instead on the one Super-Patcher tells you to flash with dumldore.

calstrin, 2019-10-20:
Thanks Brett,  I'll install this and let you know how it goes.

If I ever want to go back to a stock (non hacked) firmware is that possible?

brett8883, 2019-10-20:
Anytime. Just flash any stock firmware. You can use Super-Patcher’s “download and install stock firmware” tool (#2 in the mainmenu) or use DankDroneDownloader (option #3 in the mainmenu) to download the firmware version you want and then flash it with dumldore (#4 in the main menu)

calstrin, 2019-11-05:
Awesome thank you!

brocklong, 2019-11-12:
Hi！ Brett.
I am a P4P drone and I want to have a flight controller without Galileo.But need Smart Battery mod and stealth.
Thanks.

giannhsath, 2020-05-13:
hi brett major problem with issue on i2 after applying sp aircraft loosing gps and goes to atti mode i already do rollback in stock fw but Galileo still exists on my i2  in the beginning locks about 26-28 sats and still loosing GPS... any advice will be helpful....