brett8883/DJI_Super-Patcher #45: Inspire 2 install ADB failure

wanghuakangjq, 2019-10-16:
My Inspire 2 is now experiencing the problem of ADB failure。when I use ADB for a few seconds, the plane starts to chirp and the LED lights turn red. I try to use DUMLRacer before but its still Buzzer，how to pass ADB?

brett8883, 2019-10-17:
Copy and paste your log file to here. I assume you tried the DUMLdore “Enable ADB” button already? 

The lights changing color and chirping isn’t necessarily a bad sign, at least that means the bird is receiving the commands. On Mavic Pro the lights turn on and off and my computer chirps so I don’t necessarily know that this is out of the ordinary. 

After enabling adb with DUMLdore wait 60 seconds before proceeding. Sometimes the Inspire 2 can be slow to enable ADB. Also wait a little longer before enabling ADB after restarting. It takes Insipre 2 a little longer to fully start up than other birds.

wanghuakangjq, 2019-10-18:
thanks！ but i am sure is full start lnspire2 ,when ues ADB first is 'bi bi bi bi bi~''bi bi bi bi bi~'small sound，10seconds late front LED be red and making a harsh tweet。And i am sure is not from ESC。thanks aigan your good heart！i will try to do wait 60 seconds。

closed 2019-10-18 by wanghuakangjq