brett8883/DJI_Super-Patcher #46: Work in offline mode

Nisim620, 2019-10-19:
**thank you very very much !!!!!
MODE MOST WONDERFUL !!!

Only if you can make the tool work even without an internet connection ?!
By downloading the tool you will already have all the tools needed to operate the mode
Thank you very much**

brett8883, 2019-10-20:
This is pretty easy to achieve yourself.  You'd have to run the tool at least once online so it downloads the tools and downloads the flight controller file you want. 

Then comment out lines 151 and 152 (both start with %busybox%) of the DJI_Super-PatcherVxxx.cmd. 

Change 

`%busybox% wget https://github.com/brett8883/Super-Tools/archive/%branch%.zip && echo Download success!`

To 

`REM %busybox% wget https://github.com/brett8883/Super-Tools/archive/%branch%.zip && echo Download success!`

Change `%busybox% unzip -o -q %branch%.zip`
to
`REM %busybox% unzip -o -q %branch%.zip`

In line 92 you'd have comment out `PING -n 1 140.82.113.4 |find "TTL=" && ` to `REM PING -n 1 140.82.113.4 |find "TTL=" && `

but then you have to put `goto downloadtools` on a new line so it should look like:
`REM PING -n 1 140.82.113.4 |find "TTL=" &&`
`goto downloadtools`

brett8883, 2019-10-20:
Super-Patcher downloads dumldore and NLDapp at each run time to update to the latest versions and so I’m not hosting other people’s work. These were requirements I was given to use those tools and that’s fair enough. 

It also downloads the flight controllers dynamically so it doesn’t take up a bunch of space on your drive unnecessarily.

Nisim620, 2019-10-20:
Wow dude thank you so much for all the investment !!!!

I really thank you !!

I love people like you who help others
 It's really fun for you to know how to program it all

a great day ❤

brett8883, 2019-10-20:
Thanks you too

toadzhou, 2019-11-01:
Every time you run it, you need to download it again. Can you not compare the version after downloading it once? In the case of a new version, download the update again, because China's visit to foreign networks is really very ridiculous.