brett8883/DJI_Super-Patcher #47: inspire 2 fcc mode

giannhsath, 2019-10-23:
hello brett thanks for your good job!
is there anyway to get inspire 2 in fcc mode?

brett8883, 2019-10-24:
Officially, you can make a modified DJI Go 4 Android app with [deejayeye-modder](https://github.com/Bin4ry/deejayeye-modder) and enable the FCC patch. More info about that on [dji.retrorom.info](https://dji.retroroms.info/howto/deejayeye-modder)

Unofficially, you can try the Super-Patcher's Spark FCC mod. 

Run DJI_Super-PatcherVxxx.cmd again after having successfully Super-Patched already.  When asked which aircraft you have choose Spark. 

Choose option #5 in the main menu to run jkson fcc mod. 

Follow the prompts and restart the aircraft when instructed. 

I don't have an I2 so I can't say forsure this will help but there's some who theorize that the fcc mod for SPark will work on LIghtbridge models as well. 

Worst thing that I could see happen is it doesn't work.  If for some reason it makes thinks worst that would actually be interesting. You can remove it with Super-Patcher same way you enabled it. 

Let us know if you try this

giannhsath, 2019-10-26:
super patcher report that data transferred successfully for fcc mode ( spark model chosen ) for i2 but it doesn t work.

brett8883, 2019-10-28:
Ok thanks for the update