brett8883/DJI_Super-Patcher #48: Error getting response from the github

konradstoczynski, 2019-10-28:
I've been trying to run the super-patched on my machine, however I keep getting the following error message 

[image]

brett8883, 2019-10-29:
Is this on a PC or are you running in a VM? See if restarting your system fixes the issue. Does the script error out at that point or does it continue despite the error?

konradstoczynski, 2019-10-29:
> Is this on a PC or are you running in a VM? See if restarting your system fixes the issue. Does the script error out at that point or does it continue despite the error?

I'm running it on my PC, not in a VM. Unfortunately restarting the system didn't help fixing the issue. The script just displays the error and closes itself. I'm currently based in China, do you think that might be the reason why I get this error?

brett8883, 2019-10-29:
I’m not sure to be honest. That may have something to do with it but I’m not sure. Are using a VPN? 

SSL stuff I don’t know that much about. 

Try clicking this link and see if it will let you download Super-Tools-2.0 manually https://github.com/brett8883/Super-Tools/archive/2.0.zip

You can then put it in c:\users\[your user name]\appdata\roaming\DJI_Super-Patcher\

Hopefully you won’t have the same issue with downloading the other stuff too though

JaySCohen, 2019-10-29:
Which folder are you running superpatcher in?  Is it possible you don't have full rights to that folder?  If you run the command prompt as an admin, you don't default to a folder you have rights to.  Try creating a folder under your profile, like in your own desktop and change folders there in an administrative command prompt and see if that helps?

brett8883, 2019-11-15:
Without more information I don’t know how to move forward with this issue. I changed the download process slightly in 2.0.2 so if this continues to happen please make a new issue and I’ll look into it further. Closing this issue

closed 2019-11-15 by brett8883