brett8883/DJI_Super-Patcher #50: Mavic 2 Pro

bobster316, 2019-11-02:
Is there going to be any super patcher for mavic 2 pro ? Ive been waiting for months for mod's for this.. If its unlikely for the  mavic 2 pro to be modded, i'll have to sell it for a mavic which can be modded,....

brett8883, 2019-11-02:
> Is there going to be any super patcher for mavic 2 pro ? Ive been waiting for months for mod's for this.. If its unlikely for the mavic 2 pro to be modded, i'll have to sell it for a mavic which can be modded,....

I get this question a lot but unfortunately DJI is using an ARM chip with SafeZone on the Mavic 2. Unfortunately this is above my pay grade so if and when somebody is able to get around this then Super-Patcher will support M2 but no such method is public ally available.

bobster316, 2019-11-02:
Am I better off getting a mavic pro instead

Best Regards
Rav Khangurra

brett8883, 2019-11-15:
> Am I better off getting a mavic pro instead Best Regards Rav Khangurra
> […](#)

Yea I would

GitHub-Updates, 2020-01-21:
I would be interested in Mavic 2 Pro as well, if possible. I messed up when I installed the newest .500 firmware, now I'm stuck.

Thanks