brett8883/DJI_Super-Patcher #51: Patch Successful but NFZ still in effect?

dhoyt7141, 2019-11-04:
After a few attempts, I've finally got patching successfully complete, however the NFZ is still in effect (my house is in a NFZ). I don't see anywhere in the Patcher where NFZ is disabled or the NFZ database is cleared, is there anything additional I need to do to get NFZ disabled? I have Assistant 1.1.2 installed with debug mode working but doesn't look like a disable NFZ variable is apart of this firmware (V01.04.0300).

Thoughts?

brett8883, 2019-11-04:
> After a few attempts, I've finally got patching successfully complete, however the NFZ is still in effect (my house is in a NFZ). I don't see anywhere in the Patcher where NFZ is disabled or the NFZ [...]

brett8883, 2019-11-04:
> After a few attempts, I've finally got patching successfully complete, however the NFZ is still in effect (my house is in a NFZ). I don't see anywhere in the Patcher where NFZ is disabled or the NFZ [...]

Did you actually try to fly?

From the [readme](https://github.com/brett8883/DJI_Super-Patcher#remove-nfz-and-geozone-limits-by-default)

### Remove NFZ and GeoZone limits by default

  There is nothing additional the user needs to do to enable this feature and it is always enabled.

- As soon as the Super-Patcher process is complete the drone will ignore all forced GeoZone and NFZ limitations

  - **This is not an optional and is hard coded into the firmware.**

- Please note:

  - GeoZones will still be marked in the app and the app will still warn you when the drone is close to or in an NFZ,

  however…

    - Your drone will not be prevented from flying

    - Your drone will not be forced to auto-land.

    - You drone will not be preventing from taking off

      - **Please fly with caution!**

- The airport limit parameters are hard-coded and will not be visible in Assistant 2. Airport limits have already been disabled. These parameters have been modified to disable airport limits altogether even though they are not visible.

    **It is always the responsibility of the pilot to fly safely and to know local regulations.**

It will ignore GeoZone restrictions but both the app and the aircraft are still aware of GeoZones. It just won’t stop you from doing anything. You can use [NFZ eraser](https://github.com/brett8883/DJI_Super-NFZ_Eraser)if you wish to remove the NFZ database from the drone but in practice this doesn’t actually change what you can and cannot do.

dhoyt7141, 2019-11-04:
No, I powered on the Mavic and Remote and once the Mavic had GPS the remote said NFZ on it. In the past this has always indicated the Mavic would not arm which I why I didn't try. I will try again tonight and update accordingly, thank you!

elintnl, 2019-11-09:
After you done the NFZ eraser you can use the dji go mod 4.1.22 app, build by yourself and choose the patch "removeNFZ_ApplicationPart.patch" no more warnings from the app.
https://github.com/Bin4ry/deejayeye-modder/tree/master

brett8883, 2019-11-15:
> No, I powered on the Mavic and Remote and once the Mavic had GPS the remote said NFZ on it. In the past this has always indicated the Mavic would not arm which is why I didn't try. I will try again tonight and update accordingly, thank you!

Did you have a chance to try it?

elintnl, 2019-11-15:
Hi Brett, I will test it when I come back this weekend, I am on holliday in
Washington Alexandria Old Town.
Gr
Jos