brett8883/DJI_Super-Patcher #52: 'mavicpro_sp_2.0_battmod__dji_system.bin' it's not downloaded！How to do？

yh6895, 2019-11-05:
I can't find 'mavicpro_sp_2.0_battmod__dji_system.bin'   Because it's not downloaded！How to do？Thank !

brett8883, 2019-11-06:
Strange I’ll look at it but it’s here https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/MavicPro/FC_bin/MavicPro_SP_2.0_BattMod_03.02.44.12_dji_system.bin

yh6895, 2019-11-06:
Thank you very much for your reply！But I can't see it。Take a look at my screenshot  There's not even a raw folder。

yh6895, 2019-11-07:
I found it. Thank you very much!

brett8883, 2019-11-09:
> I found it. Thank you very much!

Glad you found it. The link above is the dissect link so clicking it will start the download process without further interaction. By default this means the file will end up in your downloads folder unless you’ve changed the location of where your files get downloaded to

brett8883, 2019-11-11:
I just tried it and it automatically downloaded for me so I am not sure what happened in your case. Maybe GitHub was down at that moment in time?

yh6895, 2019-11-11:
I don't know! I guess it's my network problem here. I download the file on your website and put it into the corresponding path to run.Fortunately, it worked.

brett8883, 2019-11-15:
Very good. I’m gonna close this issue since you got it to work and as far as I can tell it’s working as intended now. Open a new issue if anybody else has the same problem

closed 2019-11-15 by brett8883