brett8883/DJI_Super-Patcher #56: Spark Mkdir problem

tonimg87, 2019-11-11:
Hi, I tried to check this patch for a Spark with the 0900 firm but had some problems. I tried in win 7 and now im trying in win10 with same issue, when it starts connecting to the Spark by ADB the cmd write this:
--------------------------------------------
WORKING PLEASE WAIT...
adb shell mount -o remount,rw /vendor

adb shell mkdir /vendor/bin
**mkdir failed for /vendor/bin, File exists**

adb push dummy_verify.sh /vendor/bin/
dummy_verify.sh: 1 file pushed. 0.1 MB/s (1272 bytes in 0.016s)

dummy1 engaged based on verify type is Spark 1

adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd /

adb shell mount -o remount,ro /vendor

STILL WORKING... Just a sec
dummy_verify step complete
----------------------------------------------------------

I dont know where could be the problem and that is the unique error I got, the MKDIR.
The consequence is that I have the folder SP_Flight_Controllers created but any file is inside, so I cannot load new firm. 

Pleaseeeee anyone have solved this problem? 

Thank you very much

Manulu1, 2019-11-12:
by all means, You can remove the Spark mod from here, It doesnt work! more than 4 hours, follow the instructions and videos and does not do nothing!

Manulu1, 2019-11-12:
I follow all step 1 by 1, slowly reading every character and at this point Spark_SP_2.0_BattMod_03.02.43.12_dji_system IT Jumps from 20 to 129% is that even good? or like im having the problem It does not work

tonimg87, 2019-11-12:
> I follow all step 1 by 1, slowly reading every character and at this point Spark_SP_2.0_BattMod_03.02.43.12_dji_system IT Jumps from 20 to 129% is that even good? or like im having the problem It does not work

I had same issue with a file created by other Spark, but it works after that error. 

My problem is that I never got the Spark 2.0.bin file....

brett8883, 2019-11-12:
> Hi, I tried to check this patch for a Spark with the 0900 firm but had some problems. I tried in win 7 and now im trying in win10 with same issue, when it starts connecting to the Spark by ADB the [...]

That particular error means the directory that the command is trying to make already exists so it doesn't need to make it again. 

This is fine and does not affect anything at all. If you try to do Super-Patcher on a bird more than once this error will appear on the second try and every try after that. There are also some other programs that use the vendor/bin directory so even if you haven't done SP before its still possible you may already have this directory created and it still is not an issue.

tonimg87, 2019-11-12:
> That particular error means the directory that the command is trying to make already exists so it doesn't need to make it again.
>
> This is fine and does not affect anything at all. If you try to do [...]

Ok, that error I know that it is not the real problem, but why the patcher didn´t create the 2.0 Firmware file? It creates the folder SP Flight Controller but with any file inside.

Is there any log to insert here whats doing the cmd? Thank you!

brett8883, 2019-11-15:
> Ok, that error I know that it is not the real problem, but why the patcher didn´t create the 2.0 Firmware file? It creates the folder SP Flight Controller but with any file inside.
>
> Is there any log [...]

I understand now. What options did you choose for the battery mod and the stealth mod? 

I’ll take a look at it.

tonimg87, 2019-11-15:
I have tried yes/yes and no/no, with same results, I have the folder with any file inside

brett8883, 2020-02-23:
Works for me when I do it.