brett8883/DJI_Super-Patcher #57: Return To Home does not work

Makarov-JR, 2019-11-12:
hi brother, the return to home feature doesn't work in the NFZ area, is that normal?
but can still fly with GPS mode

brett8883, 2019-11-12:
Which aircraft are we talking about?

Makarov-JR, 2019-11-13:
Sorry, mavic pro bro

brett8883, 2019-11-15:
I don’t have a ton of experience flying in DJI red zones but I am somewhat surprised that it doesn’t RTH in those zones. Maybe try it again and see if you still have the same issue. I have not heard of this before. 

At any rate, I do have a remedy if it still doesn’t work. My repo [DJI_NFZ_Eraser](https://github.com/brett8883/DJI_Super-NFZ_Eraser) will clear out the NFZ database so that the drone will be unaware of any NFZ areas to begin with. This repo was recently overhauled to make it as simple and non invasive as humanly possible. It’s short easy work.

Makarov-JR, 2019-11-19:
thakyou bro, i Will try your hardwork

Makarov-JR, 2019-11-19:
Good job brother, your DJI Super-NFZ Eraser works very well, now the RTH feature can function in a place that was once the NFZ area, Thank you very much

michael-blue2, 2019-11-25:
This will have the same affect on p4p correct? But as soon as you update the zones it will have them there correct?

Makarov-JR, 2019-12-02:
I didn't know in p4p it would work because my drone is Mavic Pro 1, if you use DJI_Super-Patcher and DJI Super-NFZ Eraser then your geozone (NFZ) will no longer be updated