brett8883/DJI_Super-Patcher #58: Speed settings

jeje34980, 2019-11-13:
Hello. is it necessary to change the speed settings or the program does it itself? if the program does, what are the values ​​in sport and gps mode? thank you.

brett8883, 2019-11-14:
Super-Patcher does not change the speed settings. They remain factory default

jeje34980, 2019-11-14:
thank you brett. Do you have a link or can you give me the list of modifications made by the program? thank you for your great work.

Weirdoz, 2019-11-14:
Ever tried the readme?
https://github.com/brett8883/DJI_Super-Patcher/blob/master/readme.md

ikkezelf-drone, 2020-05-10:
speeds are set back to default factory settings after flashing the device is my observation. If you previously changed them you need to do it again