brett8883/DJI_Super-Patcher #59: error: more than one device/emulator

koenktm, 2019-11-17:
Hi There,

I was trying to run the Super patcher, but I found that the flight controller was not changing and the firmware version did not listed as 0.0.0xxx in the DJI go app.
During the process I saw the following error: more than one device/emulator

I tried running the process several times on two different computer with windows 10.

Here is the log:

START SUPER_PATCHER 
zo 17-11-2019_13:52:50,37 
I have verified internet access to GitHub 
13:52:53,78 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
error: more than one device/emulator
adb shell mkdir /vendor/bin 
error: more than one device/emulator
adb push dummy_verify.sh /vendor/bin/ 
 
dummy1 engaged based on AC type is MavicPro 1 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
error: more than one device/emulator
adb shell mount -o remount,ro /vendor 
error: more than one device/emulator
dummy_verify step complete 
Starting bind step... 
bind1 engaged based on AC type is MavicPro, verify type is 1 
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify 
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: more than one device/emulator
Start flash step 
dummy_bind complete 
Start jkson 
User requests jkson fcc mod, starting now 
jkson mod complete 
going to _verify 
_verify.cmd starting 
User Indicated patch not successful. Showing log file

koenktm, 2019-11-17:
So today I did one more attempt, while I was making this attempt my android TV was on. While the superpatcher was checking the ADB devices a promt of USB debugging appeared on my android tv. That's when I decided to take the plug out of my tv. since then the flash worked out brilliantly

brett8883, 2019-11-20:
> So today I did one more attempt, while I was making this attempt my android TV was on. While the superpatcher was checking the ADB devices a promt of USB debugging appeared on my android tv. That's when I decided to take the plug out of my tv. since then the flash worked out brilliantly

Yep that’ll do it. Would have guessed it was your phone but tv will do it too if it runs on Android. 

Basically ADB didn’t know which device you wanted to send the commands to. By unplugging the tv there was no longer a doubt and so it sent the commands successfully.

mmaacckkii, 2021-10-01:
> So today I did one more attempt, while I was making this attempt my android TV was on. While the superpatcher was checking the ADB devices a promt of USB debugging appeared on my android tv. That's when I decided to take the plug out of my tv. since then the flash worked out brilliantly

Hi i have the same problem but i do tego have Connected aby other devices to USB, i have only android tv connected to my WiFi.