brett8883/DJI_Super-Patcher #61: 2.0.2 with Mavic pro, custom FW upload hangs

Mussemaan, 2019-11-19:
No errors, no apparent reason.
Original xxx.300 FW goes in just fine in 6 minutes. But ran out of about 50% battery in first time. Seems that second time will not solve it either. DUMLdore has been 100% and "transferring to device" now 15-20 min and still counting.. Win10, procedure followed to the letter.

Mussemaan, 2019-11-19:
Tried first time with battery mod activated. Now in 2nd try and again xxxxx.300 went in smoothly first.

Both battery and stealth mode option in beginning selected to "No".  Enables ADB fine, no errors again. Just sticks to 100% and "Transferring to device". 20min now gone, about 17min with 100% green bar. 50% of battery remaining, started with about 75% full this time.

brett8883, 2019-11-20:
> Tried first time with battery mod activated. Now in 2nd try and again xxxxx.300 went in smoothly first.
> 
> Both battery and stealth mode option in beginning selected to "No". Enables ADB fine, no errors again. Just sticks to 100% and "Transferring to device". 20min now gone, about 17min with 100% green bar. 50% of battery remaining, started with about 75% full this time. _Does not go into "Flashing in progress xxx %"_, hangs to transferring step.*
> 
> Mavic has not rebooted itself, apparently it should do it too once before final manual reboot.

If the drone has less than 50% battery it will not accept a firmware flash and will hang indefinitely at the transferring stage. If it doesn’t start flashing after about 10 seconds it’s not going to and you can close dumldore and restart the drone and then redo the Super-Patcher process. 

If you have already successfully flashed the stock firmware twice before beginning the Super-Patcher process then it’s not necessary to do it again. However, getting back to back successful flashes of the stock firmware before doing Super-Patcher is important. This isn't required in all circumstances, however, it is absolutely required in certain circumstances and can’t hurt so that’s why I say to do it twice always before beginning.

Mussemaan, 2019-11-20:
Thanks. Re-flashed back to original xx.300 with patcher, not DJI assistant if it matters.

- 1st time freezing could well be due to 50% or less battery
- 2nd time I had 75-80% when patching started so likely its something else causing it

Either way, FCC mod works like a charm. Thanks plenty - donation coming soon :)