brett8883/DJI_Super-Patcher #65: Aeroscope off??

nodojh, 2019-11-21:
Is probably off aeroscope?

brett8883, 2019-11-23:
This isn’t a feature of Super-Patcher

mitomenai, 2019-12-19:
I'd like to 2nd this feature request as well. I wouldn't mind if telemetry was only readable via DJI equipment sold to law inforcement, but as it turns out, anyone with a laptop and Kismet can easily access and see the real-time location of the pilot.

https://i.imgur.com/bv8zuAX.png

To make matters even worse, DJI's latest move is to make Aeroscope accessible to the public via a smartphone app. This is an absolute disaster.

https://www.dji.com/newsroom/news/dji-demonstrates-direct-drone-to-phone-remote-identification

Some people can be very irrational and violent when it comes to drones, and now anyone will be able to download an Android app that spits out a nice Google Maps picture pointing them to the exact coordinates of the pilot - no need for Linux knowledge or Kismet.

Is there any possible way of disabling this beacon? I'd even be willing to donate $200 for your effort.