brett8883/DJI_Super-Patcher #66: I follow all step 1 by 1, slowly reading every character and at this point Spark_SP_2.0_BattMod_03.02.43.12_dji_system IT Jumps from 20 to 129% is that even good? or like im having the problem It does not work

SkyNaras, 2019-11-22:
I follow all step 1 by 1, slowly reading every character and at this point Spark_SP_2.0_BattMod_03.02.43.12_dji_system IT Jumps from 20 to 129% is that even good? or like im having the problem It does not work

_Originally posted by @Manulu1 in https://github.com/brett8883/DJI_Super-Patcher/issues/56#issuecomment-552708999_

brett8883, 2019-11-23:
> I follow all step 1 by 1, slowly reading every character and at this point Spark_SP_2.0_BattMod_03.02.43.12_dji_system IT Jumps from 20 to 129% is that even good? or like im having the problem It does not work
> 
> _Originally posted by @Manulu1 in [#56 (comment)](https://github.com/brett8883/DJI_Super-Patcher/issues/56#issuecomment-552708999)_

Yes it’s fine and expected. Per the program:
—-
 1. Please click "Load Firmware" button in DUMLdore

2. Choose the file %fw2% it is in your SP_Flight_Controllers folder which is at:
 %FCfolderpath%

3. Then click "Flash Firmware" in dumldore

4. Please allow DUMLdore to flash to the aircraft.

## NOTE: Progress bar in DUMLdore is not accurate and may go above 100^% or may seem to hang at 100% for some time but will complete

SkyNaras, 2019-11-23:
But it happens within seconds and the DUMLdore closes

omegafritz, 2019-12-01:
Yeah same situation happens with my Spark. I then wait up to 15 minutes manually restart and when I go to DJI GO 4 app firmware still says 1.000.900...However assistant 2 says its on 0.000.000 and when I attempted to fly it it seemed to have worked. Is it possible the dji go app isn't showing the firmware correctly?
[log.txt](https://github.com/brett8883/DJI_Super-Patcher/files/3908687/log.txt)

planetzog, 2020-04-19:
I'm having problems flashing too (reported above) but noted your comments about Assistant 2. Mine shows Version 00.00.000 too, but is stuck on CE mode (no FCC) so I guess it hasn't worked.