brett8883/DJI_Super-Patcher #67: DUMLdore crash when loading custom firmware - System.ArgumentOutOfRangeException

dol1234, 2019-11-23:
Hi,
First of all thank you for all work you've done:)
Trying to patch my Spark, was able to patch with V01.00.0900_Spark_dji_system.bin twice, but when loading Spark_SP_2.0_BattMod_03.02.43.12_dji_system.bin DUMLdore crash. In windows event logs I can see info about uncaught  exception (you can find exact message in polish below).
I assume this might be connected with update progress larger than 100%. Is there something that can be done in this case?

```
Aplikacja: DUMLdoreV3.exe
Wersja architektury: v4.0.30319
Opis: proces został przerwany z powodu nieobsłużonego wyjątku.
Informacje o wyjątku: System.ArgumentOutOfRangeException
   w System.Windows.Forms.Control.MarshaledInvoke(System.Windows.Forms.Control, System.Delegate, System.Object[], Boolean)
   w System.Windows.Forms.Control.Invoke(System.Delegate, System.Object[])
   w DUMLdoreV3.ControlExtensions.SafeInvoke[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]](System.Windows.Forms.Control, System.Func`1<Int32>)
   w DUMLdoreV3.Form1.UpgradeStatus(Duml.Msg)
   w Duml.MsgParser.HandleIncomingMsg(Duml.Msg)
   w Duml.MsgParser.ReadFromPort(System.Object)
   w System.Threading.ThreadHelper.ThreadStart_Context(System.Object)
   w System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   w System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   w System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   w System.Threading.ThreadHelper.ThreadStart(System.Object)
```

brett8883, 2019-11-23:
Post your Super-Patcher log please

dol1234, 2019-11-24:
Actually nothing there, please check below.
Same issue on two different machines (both Win10). Tested with different cables, three different batteries (fully charged). 
Tried also Spark_SP_2.0_FullyLoaded_03.02.43.11_dji_system.bin with same result.
When flashing progress bar goes to 126%, then DUMLdore freeze abd crash without error message, Drone chirped few times, restarted twice and looks ok but I have no idea if DUMLdore does any work after software is uploaded to device or not.
After crash I was able to restore to stock firmware (during first restore there is multiple chirp on 26% and fan goes down). 

`START SUPER_PATCHER 
24.11.2019_ 7:24:00,25 
I have verified internet access to GitHub 
 7:29:16,45 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
adb shell mkdir /vendor/bin 
adb push dummy_verify.sh /vendor/bin/ 
 
dummy1 engaged based on AC type is Spark 1 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
adb shell mount -o remount,ro /vendor 
dummy_verify step complete 
Starting bind step... 
bind1 engaged based on AC type is Spark, verify type is 1 
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify 
* daemon not running; starting now at tcp:5037
* daemon started successfully
Start flash step 
`

brett8883, 2019-11-26:
> Actually nothing there, please check below.
> Same issue on two different machines (both Win10). Tested with different cables, three different batteries (fully charged).
> Tried also [...]

Ah. I’m not a Spark owner but I am told this happens with Spark sometimes. The patch is successful just allow it to do its thing and when it crashes give it a little more time to finish the process and then restart the drone and tryout should be good to go. 

The progress bar going over 100% is a good indication that the patch is successfully being applied. I’m not sure why this happens with Spark and not the others but it’s fine.

dol1234, 2019-11-26:
Can confirm that despite the error Spark was upgraded. 
Works great (including FCC mod). Thank you:)

brett8883, 2019-11-26:
Reopening issue because It is good for other people who may be experiencing the same thing

Tonic0, 2020-01-20:
2.0.2 doesn't seem to be flashing to Spark, DUMLdore exits without any messages after around 20% progress once initiating the firmware flash stage. I've let the flash finish and restarted the Spark after it has auto-rebooted but it's still sitting on stock firmware. Tried double flashing stock FW on both the tool and NLD program but no luck.
Used 1.7.5 previously and that worked a charm. Not sure if a W10 update since then has broken the process or what's going on.

planetzog, 2020-04-19:
I'm having the same problem. Crashes when DUMBLdore shows 125%, then a Windows message, "This program has closed." I've tried two different versions of Windows 10, double flashed the required version, and waited in case it's working anyway. It wasn't. Version stills shows as unchanged at 01.00.0900. Tried multiple times, turned off virus checker, nogo. Any ideas?