brett8883/DJI_Super-Patcher #70: Question about NFZ removal on P4P2

StanLee4456, 2019-12-09:
I have an opportunity to purchase a P4P2 on latest fw. If I do, I will run SP to remove alt/NFZ. Does SP also work to adjust parameters like increasing the ascent/descent speed?

Also, when NFZ is removed, how does it work? Do NFZs still show in GO4 and you just don't need to "unlock" them?

Thanks in advance - I just want to find out how exactly this works before parting with my money. Thanks again!

brett8883, 2019-12-24:
Super NFZ Eraser has not been tested on P4PV2, I do not know if it will work so you’d be doing it at your own risk. To be clear, however, that process is likely redundant. 

On P4PV2 there is an NFZ parameter that is hidden in the firmware. By default it is switched to on. Receiving and unlocking license from DJI switches this parameter to off. 

Super-Patcher switches this parameter to off by default. This was enough for all the other aircraft to bypass NFZ restrictions, however, not on the P4PV2. It seems that the app was able to switch this parameter back on once the app was connected to the remote and the remote was connected to the aircraft. This is why disabling NFZ’s on P4PV2 didn’t work on early versions of Super-Patcher. 

In recent versions on Super-Patcher, at the advice of someone whom I forget their name off the top, instead of simply switching the parameter to the off position I actually removed the ON position altogether. The app can no longer switch the NFZ parameter back on because it is now static and hard coded into the firmware.

StanLee4456, 2019-12-25:
Brett - can you write to me privately on the dii-rev forum? I can't pm here. I have some ideas.
StanLee.