brett8883/DJI_Super-Patcher #71: Spark worked great-- a couple questions-observations

duecedriver, 2019-12-10:
First off let me say great work.. bundling and scripting all the various pieces made this process worth doing finally.. I have a Spark and a Mavic and decided to try the spark first...

First I had to downgrade from .1000 to .900  I only did the aircraft and the battery that was in it.. left the other batteries alone and have gotten no warnings so I assume no harm - no foul ... also, the controller is still at what .600 I think?  whatever it would have been for .1000 firmware.. any issues with this or is it OK to keep the controller at the latest firmware?

The patching itself went pretty good, I downloaded the stock .900 firmware from within the 'modder' but it would not install it, giving me an error that it could not connect.  I had to close out of everything and start it backup up. still nothing so ran dumldore on its own and flashed it.

during the 'scripted' mod install, one of the steps where the instructions were to leave it to reboot  several times and not reboot manually until dumldore says its OK.. seemed like the window closed/crashed out.. waited and waited but nothing so decided to shut the drone down and reboot and it moved on and reported firmware .0000  

I chose the battery mod, and the boost mod

seems like the battery mod really cuts it close on the RTH return point.. don't be flying back into the wind you wont make it.!!!  as part of the FAQ you mention that Assistant 2 can tune/reset these values.. would you mind setting up a FAQ that has the descriptors to tune for each of the mods so us NOOBS know which knobs to turn!! hehe

I realize that these mods require operators to put on the big boy pants.. but can you tell someone who has no experience with race quads.. what battery voltages are 'safe' for the spark/mavic before random shutdown/failure/ fall out of the sky becomes likely ?

can't really tell if the boost mod is working on the spark, I am in FCC already so no help there and I keep it in auto so here it defaults most of the time to the 5.8 band.. I can get out about 4-5000 ft before signal loss.. not a noticeable improvement that I can tell yet but my area has lots of wifi interference ..  Q.  is it possible leaving the controller at firmware .600 could be affecting this mod?  

really great work again!! much appreciated!!

misterpeppery, 2019-12-10:
I flashed on 2.0.1 with battery and FCC boost mods and had similar experiences to yours. I tried reflashing several times from different computers (win7 pro 64 and win10 home 64) and could never see much increase in range. I flashed again with the stealth mod also enabled and my range and video transmission increased quite a bit. In areas where I could go 5000' I can now go 8-10k, and in areas where I was struggling to get past 1200' cleanly I can now go 3000'. I can't say if it was just that last time flashing that did the trick or if there is a difference in the patcher with the stealth mod version, but you should be seeing a noticeable increase in range of around double. If you are not seeing that I would try patching again until you do.

I did not do any other changes to transmitter or battery firmware, only the Spark itself. I was fully up to date with everything before downgrading the Spark and then patching and everything has been working fine for me.

You can follow the link in the STEALTH MOD section of the patcher notes for a more detailed walkthrouh about parameter mods using DJI Assistant, but it doesn't go into any detail about the low battery tuning. I was flying my Spark the other night just to see how long I could stay up after the battery hit 0% and I was surprised at how long I could keep flying before it forced a landing. I wasn't very high up and was doing this over grass just in case, so I don't know how high it would have battery to come down from after it quits taking vertical inputs from the controller, but I'll bet it stayed flying for close to 2m after reaching 0% so I wouldn't worry too much about the new battery tuning.

duecedriver, 2019-12-11:
@misterpeppery

I have no interest in enabling stealth mode... turning off the lights...

what I might do is try flashing the jskson power mod using dumldore directly 

I did the patcher on a mavic last night.. updated firmware from .100 to .300 from inside super patcher then ran the same mods.. BATT and JKSON Auto/Looped and that seemed to work better.. much more noticeable gain in signal strength .. residential area with lots of WIFI and was out at 1.5M with little hiccups ... 

really the nicest thing with this mod is just the inclusion of the Galleleo constellation .. a nice warm fuzzy flying with 20+ sats almost all the time now.. 

hopefully there are no negatives to this approach.. I can think of a couple..

1> DJI while reporting the sats can only use XX number of sats at any given time and it does not pick the sats with the lowest overall H/P DOP loss .. i.e. doesn't take into account geometry 

2>  that having this many sats in tracking if all are being used slows down the processor or in some way reduces performance instead at the expense of enhanced positional accuracy

I have only done 2 or 3 flights on both the spark and mavic at this point post mod.. but so far no bad behaviors.. no disconnects, fly-away attempts etc..

only mod I would make to the JKSON mod would be to have a  version of it for people that are already in FCC areas that don't have that as the 'first cycle' in the list.. 

IE cycling once into FCC mode while already in FCC mode is redundant .. would prefer to just have Normal>boot>recycle

and honestly probably a safer option would be to not have it cycle past boost.. IE once I am in boost.. why would I want to cycle back to off, and making the drone reboot and RTH.. its a recipe for a possible disaster if one gets lost in the loop and accidentally cycles past Boost with the drone in a bad position or has a flyaway on the RTH ...

misterpeppery, 2019-12-12:
I tried flashing the fcc mod directly, as well as tried using a script I found in the comments here on GitHub. Neither worked for me. The stealth mode does not turn off all the lights. It turns off the rear lights, but the front red ones stay on -- unless you are recording video, during which they turn off. I did not want to do the stealth mode either, but having the better range was worth the trade-off of losing the rear lights. Maybe someday I will re-flash to get my lights back, assuming it can be determined (and fixed) that there is an error in the non-stealth-mod code.

jerryariass, 2020-03-20:
hola soy Colombiano, he probado el superpatcher en un dji spark, he tenido muy buenos resultados pero siempre alerta a cualquier falla o mal comportamiento, he probado todo los modos del firmware STEALTH, BATT MODE, NORMAL MODE, FORCE FCC+TX POWER..
hasta ahora mis resultados han sido positivos, inicialmente mi problema era el limite de altura y NFZ, con lo cual quede muy complacido, el vuelo nocturno sin luces tambien me funciono perfecto solo que asusta no ver nada en el cielo a lo lejos como era costumbre, el mod de la bateria tambien trabaja perfecto solo que no me convence mucho tener esa responsabilidad en la mente mientras se vuela, el force fcc tambien me funciono muy bien, logre mucha mas distancia en el primer test de vuelo, solo tengo una duda al respecto y es que cuando vuelo la transmision de video se ve entre cortada, es decir no es tan fluida se sienten micro lags, pero al probar todo esto decidi flashear de nuevo desde cero en el firmware 0.0900 y usar el  superpatcher sin mod de bateria ni el modo sigilo y tampoco el force fcc, de resto muy complacido, quedo muy sorprendido con los resultados y si alguien experimenta mejores resultados con la imagen en la version full porfavor compartan la experiencia, muchas gracias a sus desarrolladores y SALUDOS DESDE MEDELLIN COLOMBIA!

duecedriver, 2020-06-22:
i have not flown the spark in months, noticed today that it made no attempts at precision landing?

did multiple take offs going straight up to 40ft before flying out.. it will RTH, decent, not make any attempt at precision position adj, hover at 1ft and confirm the landing on screen?