brett8883/DJI_Super-Patcher #72: Firmware Check

racnigel, 2019-12-11:
Hi Brett,
After flashing im checking the firmware version in the go 4 app and it is saying version check collecting firmware version continuously and  when i go to about it says "check for updates" but does not show 00.00.0000 is this right? on version 4.3.24

brett8883, 2019-12-17:
> Hi,
> After flashing mavic pro im checking the firmware version in the go 4 app and it is saying version check collecting firmware version continuously and when i go to about it says "check for updates" but does not show 00.00.0000 is this right? on version 4.3.24 . does it need a different version of go4?
> PS. ive flashed it again for the umpteenth time even in dji assist 2112 it shows looking for fimware continuously in dex no other setting show?

Assistant 2 1.1.2 isn’t capable of of reading the firmware version correctly on the Mavic Pro as that’s not the version that Mavic Pro is designed to work with. DJI Go 4 usually will pick up the firmware version as 00.00.0000 but I have heard from some people that on specific devices or versions of DJI Go 4 it may not show the 00.00.0000 but instead shows nothing. They report that it works just fine. 

You can fly in the simulator of Assistant 2 to confirm working order of the flight controller. If there was a problem it would show up in the sim. 

Note that the SIM is only 500m talk so it can’t be used to confirm the height limit has been defeated but just to confirm normal flight readiness of the flight controller.

racnigel, 2019-12-17:
Thanks Brett,I have flown it since and everything seems great thanks for your great work. Nige.Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/brett8883/DJI_Super-Patcher/issues/72?email_source=notifications\u0026email_token=AKCAKCF4OU6TW5T6LCYUIWDQZARLPA5CNFSM4JZG5KNKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHAXV4A#issuecomment-566328048",

"url": "https://github.com/brett8883/DJI_Super-Patcher/issues/72?email_source=notifications\u0026email_token=AKCAKCF4OU6TW5T6LCYUIWDQZARLPA5CNFSM4JZG5KNKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHAXV4A#issuecomment-566328048",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]