brett8883/DJI_Super-Patcher #73: SP doesn't support latest fw P4Pv2?

StanLee4456, 2019-12-20:
Trying to use SP on latest P4Pv2 fw. Not working - seems to downgrade the fw to an old one..

brett8883, 2019-12-20:
The .2200 firmware isn’t listed as a compatible firmware. You can downgrade to a comparable firmware to do the patch. 

However, we were playing around with the .2200 firmware before I started my new job and now don’t have as much time to tinker. Those who have tried it it has worked. You need the correct file to flash though. 

https://github.com/brett8883/Super-Firmware_Cache/blob/2.0/P4Pv2/FW_2200/P4Pv2_2200_updated_fixed_nfz_03.03.13.99_dji_system.bin

It’s the same process but when you get the part where it tells you to flash a file with DUMLdore, flash this file instead.

StanLee4456, 2019-12-20:
Perfect. Worked like a charm. Any chance you could make it where it won’t always warn you that there is a new fw to upgrade to?

brett8883, 2019-12-24:
Just to clarify for everyone, did you use the file I linked you above to do Super-Patcher on the .2200 firmware or did you downgrade to .1500?

StanLee4456, 2019-12-25:
I used your file - worked great! Hopefully you can put a new file up for today's fw.