brett8883/DJI_Super-Patcher #74: New firmware today

StanLee4456, 2019-12-25:
Hey Brett- there was a new firmware today to improve some camera settings on the P4P2. 

Can you upload another file to make this compatible? As a Xmas gift?
Thanks so much.

knarfboy, 2019-12-26:
the v01.00.50.00 has a anti rollback trap

StanLee4456, 2019-12-26:
True. But as long as SP removes the alt + NFZ restrictions I don’t care.

knarfboy, 2019-12-26:
@ Brett8883,
I see that you have little time, but I would like to ask you to patch a current bin file for the P4Pv2.0 for us.
Merry Christmas

StanLee4456, 2019-12-27:
Brett- you still around these parts?

knarfboy, 2019-12-31:
digdat0 made it:
https://github.com/digdat0/DJI_FC_Patcher/blob/master/FC_patch_sequence_for_dummy_verify.sh

StanLee4456, 2019-12-31:
> digdat0 made it:
> https://github.com/digdat0/DJI_FC_Patcher/blob/master/FC_patch_sequence_for_dummy_verify.sh

I tried using that as the firmware with superpatcher. It doesn’t seem to work.

knarfboy, 2019-12-31:
dummy_verify is not a patched Firmware!

https://dji.retroroms.info/howto/manually_installing_custom_fw_from_fc_patcher

StanLee4456, 2019-12-31:
Sorry- he had sent me a bin file. It didn’t work.

StanLee4456, 2019-12-31:
> dummy_verify is not a patched Firmware!
> https://dji.retroroms.info/howto/fc_patcher_tool
> https://dji.retroroms.info/howto/manually_installing_custom_fw_from_fc_patcher
> 
> Best wait for Brett8883 update super patcher...

Hopefully it won’t take too long. Maybe a New Year’s gift?

knarfboy, 2019-12-31:
what bin file he had send to you?

StanLee4456, 2019-12-31:
https://drive.google.com/file/d/1ZlX9QGG18axYVZCDoEbo1L1ZCMBw3eFx/view?usp=sharing

StanLee4456, 2019-12-31:
But the bin file doesn’t work.

knarfboy, 2020-01-02:
@brett8883 ,
can you create a v01.00.50.00 bin file? I will test it.

StanLee4456, 2020-01-03:
@brett8883 ,
I’ll happily do testing as well.

knarfboy, 2020-01-06:
If P4P2.0 is upgraded to 01.00.50.00 firmware, it has rejected dumldore's adb function.