brett8883/DJI_Super-Patcher #75: Problems with P4PV2

Dronesafe, 2019-12-30:
Hi brett,
When I completed all steps on my P4PV2 with Super-Patcher1.9, the NLD app showed correct flight controller number，but the status LEDs on my AC were flashing Red and Green alternately. Still I could start the motors and there were no errors occured on DJI Go 4 app. I don't know what was wrong with my work, could you help me?

Dronesafe, 2019-12-30:
Hi brett,
Also I tried Super-Patcher 2.02 on my P4PV2, the status LEDs on my AC were Red flashing, but everyting was OK on DJI Go 4 app. I got into trouble with LEDs，T.T(crying).

knarfboy, 2019-12-30:
@dronesave,
if you have Firmware 01.00.22.00 or less try:
https://github.com/brett8883/Super-Firmware_Cache/blob/2.0/P4Pv2/FW_2200/P4Pv2_2200_updated_fixed_nfz_03.03.13.99_dji_system.bin
It’s the same process but when you get the part where it tells you to flash a file with DUMLdore, flash this file instead.
in case you have Firmware 01.00.50.00 you have no luck. DJI fix some and add a antiorollback trap

Dronesafe, 2019-12-31:
@knarfboy Thanks a lot. So which version of Super-Patcher is more compatible with P4PV2, v1.9 or v2.02, or both OK?

knarfboy, 2019-12-31:
use v2.0.2

digdat0 already hack the new Firmware v01.00.50.00
https://github.com/digdat0/DJI_FC_Patcher/blob/master/FC_patch_sequence_for_dummy_verify.sh

I expect brett8883 will update superpatcher soon

StanLee4456, 2019-12-31:
Hopefully. Because the current superpatcher doesn’t work with the latest fw -even using digdat’s file as the firmware to flash.

Dronesafe, 2020-01-01:
@knarfboy Sounds great！Thanks for your help, LEDs are flashing nomally now, and the patcher works well on P4PV2. Let's wait for the new version of SP.