brett8883/DJI_Super-Patcher #78: HELP please.. Cant patch spark, "no devices/emulators found" error and stuck in .900

sirblackhand, 2020-01-25:
hello, first saying im using old pc thats all i have avaliable, and probably the error comes from there. i follow all the steps but no chance for this to work.
i saw in the read.me that when not using win10 can be insstalled something manually to avoid this error, but i cant find anything more about it, so no clue what to look for..

heres the log from super patcher

> START SUPER_PATCHER 
25-01-2020_ 5:46:08,81 
I have verified internet access to GitHub 
 5:46:19,00 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
error: no devices/emulators found
adb shell mkdir /vendor/bin 
error: no devices/emulators found
adb push dummy_verify.sh /vendor/bin/ 
 
dummy1 engaged based on AC type is Spark 1 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
error: no devices/emulators found
adb shell mount -o remount,ro /vendor 
error: no devices/emulators found
dummy_verify step complete 
Starting bind step... 
bind1 engaged based on AC type is Spark, verify type is 1 
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify 
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found
Start flash step 
dummy_bind complete 
Start jkson 
User declined FCC mod 
going to _verify 
_verify.cmd starting 
User Indicated patch not successful. Showing log file 
Start jksonMM 
User requests jkson fcc mod, starting now 
jkson mod complete

brett8883, 2020-01-25:
Yes the error you are having is typically the issue reported when using windows 7 or 8. 

The adb drivers from DJI don’t seem to work on Windows 7 or 8. I can’t provide additional support for those versions of Windows but it has to do with the DJI adb driver.

sirblackhand, 2020-01-25:
> Yes the error you are having is typically the issue reported when using windows 7 or 8.
> 
> The adb drivers from DJI don’t seem to work on Windows 7 or 8. I can’t provide additional support for those versions of Windows but it has to do with the DJI adb driver.

Thanks for the reply. 
So only solution is trying in a win10 pc?

brett8883, 2020-01-26:
> Thanks for the reply.
> So only solution is trying in a win10 pc?

Thats the only solution I am able to offer. Maybe some one will have another idea and I know there have been people that have successfully used older versions of Windows but I haven’t been able able to isolate why they have been able to get it to work. I only have Windows 10 so I don’t have a way to do my own research.

sirblackhand, 2020-01-26:
> Thats the only solution I am able to offer. Maybe some one will have another idea and I know there have been people that have successfully used older versions of Windows but I haven’t been able able [...]

Hey, I think I did it!
I could install win10, and follow the steps. During the flash dumlore went to those famous 126% and closed, but the spark reboot and chirped three or four times and after the front leds went blinking green. I gave it some time more and continued with the process.
The thing is when I go to dji go4 the version of AC appear as 'N/A' and not those 0.000.00
Is this signal of fail? Or the flash worked? 
I can't find any other way to check it. 

About the fcc mod I already had the fake location thing, so I still get the message to change wlan (the one to press Cancel) when I connect.

brett8883, 2020-01-27:
> Hey, I think I did it!
> I could install win10, and follow the steps. During the flash dumlore went to those famous 126% and closed, but the spark reboot and chirped three or four times and after the [...]

Good to hear! The n/a is good. Will update at some point to reflect that some versions of the android app now show n/a.

sirblackhand, 2020-01-27:
Well if the N/A is normal then everything is ok.
I could fly a little to try and after a while the # of satellites went to 22/23. And could test the POI and cinematic mode and both worked well. 
About distance, didn't went further than 500m because was night and was first time flying outside with new firmware and new app, and didn't wanted to take too much risk. But I could notice a big difference in the video feed and not even once I lost image or a flinch etc.
So I think everything is working fine and the problem was solved. 
Thanks lot for all your help

brett8883, 2020-01-27:
> Well if the N/A is normal then everything is ok.
> I could fly a little to try and after a while the # of satellites went to 22/23. And could test the POI and cinematic mode and both worked well. [...]

My pleasure