brett8883/DJI_Super-Patcher #79: No 00.00.0000 dji go app

rtvater, 2020-02-02:
I am having the same issue as well I have updated all my driver's on the PC flash it 20 million time and still  no 00.00.0000 could it be the abd drivers on window 10 if any one out the problem please let me no dji phantom 4 pro

brett8883, 2020-02-04:
Please copy and paste the contents of your log file to this issue. 

What does DJI Go 4 say your firmware version is for the aircraft?

rtvater, 2020-02-05:
did you get it

brett8883, 2020-02-05:
Did I get what?

rtvater, 2020-02-05:
my log

rtvater, 2020-02-05:
START SUPER_PATCHER 
Fri 01/31/2020_12:13:19.05 
I have verified internet access to GitHub 
12:13:21.74 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
adb shell mkdir /vendor/bin 
adb push dummy_verify.sh /vendor/bin/ 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /system/bin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
dummy2 engaged based on AC type is P4 verify type is 2 
adb shell mount -o remount,ro /vendor 
dummy_verify step complete 
Starting bind step... 
bind2 engaged based on AC type is P4, verify type is 2 
* daemon not running; starting now at tcp:5037
* daemon started successfully
Start flash step 
dummy_bind complete 
Start jkson 
going to _verify 
_verify.cmd starting

rtvater, 2020-02-05:
here it is

closed 2020-02-05 by rtvater