brett8883/DJI_Super-Patcher #81: no 00.00

rtvater, 2020-02-05:


rtvater, 2020-02-05:
START SUPER_PATCHER 
Fri 01/31/2020_12:13:19.05 
I have verified internet access to GitHub 
12:13:21.74 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
adb shell mkdir /vendor/bin 
adb push dummy_verify.sh /vendor/bin/ 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /system/bin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
dummy2 engaged based on AC type is P4 verify type is 2 
adb shell mount -o remount,ro /vendor 
dummy_verify step complete 
Starting bind step... 
bind2 engaged based on AC type is P4, verify type is 2 
* daemon not running; starting now at tcp:5037
* daemon started successfully
Start flash step 
dummy_bind complete 
Start jkson 
going to _verify 
_verify.cmd starting 

is this what you looking for

brett8883, 2020-02-08:
Looks fine to me. Go and test it

closed 2020-02-16 by brett8883