brett8883/DJI_Super-Patcher #82: P4P focus

seric93220, 2020-02-06:
hi all, what's up !
i have a P4P with crystalsky ( CS ) and 4HAWKS raptor XR antenna and i'm just looking for power , because i'm stuffed with loosing connection behind the vegetable garden hedge at 100 meters... :( 
So ! questions with superpatcher :
1/ a have read ( here ) that P4P has no FCC mode ? 
2/ the same for BOOST ? does BOOST just burn transmitter (heat) ?
3/ does i can use last DJIGO4 with my CS, it mean's just apply super patcher on my bird and CHANGE NOTHING with CS ?
4/ does super patcher downgrade transmitter in the good firmware ?
5/ finally, sorry for that guy's, always fly with DJIGO4 in plane mode ?
thank's to brett, i'm so impressed !

brett8883, 2020-02-08:
> hi all, what's up !
> i have a P4P with crystalsky ( CS ) and 4HAWKS raptor XR antenna and i'm just looking for power , because i'm stuffed with loosing connection behind the vegetable garden hedge at [...]

Thanks! 

The Jkson FCC mod supports *P4P V2* but not P4P. No FCC nor Boost mode modification on P4P with Super-Patcher. For Mavic Pro, P4P V2, and Spark boost mode doesn’t cause overheating. 

Boosting the P4P signal can be done by adding additional external amplifiers which is easy to do since you have already modified the RC to install the antennas, however, this is outside of the scope of the project. 

Super-Patcher is an aircraft firmware modification. It will work with any version of the DJI Go 4 app because it does not rely on the app to work and does not interfere with app’s functions. Super-Patcher doesn’t 

You do not need to do anything to your RC or CS. No downgrading required. 

I always fly in airplane mode because I think it improves RC link quality but this is not required.

seric93220, 2020-02-08:
ok, i have seen for maxxuav booster of course but... what a mess !! It look's like P4P is blocked for fcc mod or boost. sniff... thank's brett for your answer.

brett8883, 2020-03-19:
You’d need two of these 

Sunhans eSunRC RC WiFi Signal Booster, 5.8GHz and 2.4GHz Dual Band 2000mW

https://www.amazon.com/dp/B073GS2QCQ/ref=cm_sw_r_cp_api_i_6pVCEb35HBR3A

seric93220, 2020-03-19:
I prefer this one, or selling my P4P for V2…

 

https://www.maxxuav.com/collections/dji-phantom-4-pro/products/dji-evolite-kit-p4p-p4p

 

thank’s for your mail. Taka care….

 

https://www.amazon.com/dp/B073GS2QCQ/ref=cm_sw_r_cp_api_i_6pVCEb35HBR3A