brett8883/DJI_Super-Patcher #83: M1 disarm left stick full down after SP

KlapanBig, 2020-02-09:
Hi
Before takeoff  my M1 disarm left stick full down
Standart disarm works too

brett8883, 2020-02-13:
I can’t see whatever you have linked so I have no way to comment on it

brett8883, 2020-02-13:
I can’t imagine how it could be described as dangerous.

KlapanBig, 2020-02-13:
https://t.me/djifaqchannel last messedge . 
google translate from russian :   
Important warning❗️
Super Patcher can lead to inappropriate drone behavior, spontaneous dysarma in flight, uncontrolled climb due to the fact that the inertial measuring device is not interrogated after takeoff! Crashes are not unique! To remove the patcher, it is enough to reflash the drone with the original firmware. The method described on the site https://dji.maximlord.ru is completely safe and free from this problem.

brett8883, 2020-02-16:
No, none of that is true.

brett8883, 2020-02-16:
maximlord suggests the same process. He just lists out the manual commands to install the firmware patch where Super-Patcher is an all in one script. The commands are the same

Kaktusoff, 2020-02-16:
I can approve that the firmware, which superpatcher makes didn't work properly on p4p v2.0. After flashing drone couldn't catch any satellites. But @maximlord 's firmware works perfectly! But there are no ability to turn off the LEDs.