brett8883/DJI_Super-Patcher #84: Motor current error

MSkjel, 2020-02-10:
Hi, I keep getting motor current errors on my Mavic Pro Platinum. This also happens on V01.04.0300 stock. But it doesnt happen when I update to V01.04.0400 or .500. I read in the documentation that V01.04.0300 should have the correct ESC modules. Is this something we can do anything about or is it just my Mavic?

brett8883, 2020-02-13:
I’ve never had this issue my my Platinum and haven’t heard anyone having erroneous motor current errors on 01.04.0300. However there is an ESC update in 01.04.0400 so it is a different module. I haven’t noticed any real flight difference but the start up chime is noticeably different. 

You can do SP on 01.04.0300 and then once it’s done you can use this firmware package to upgrade to 01.04.0500 and keep the Super-Patcher mods. Just flash this firmware with dumldore after successful completion of SP on 01.04.0300. 

https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/MavicPro/Upgrade_01.04.0300_to_01.04.0500/Upgrade_Custom_FC_MavicPro_to_01.04.0500_NO_305_NO_306_NO_905_NO_801_dji_system.bin

Note that after flashing this file DJI Go 4 and Assistant 2 will again show the firmware version correctly as 01.04.0500 so there’s no way to verify the mods remain but the mods will remain. You can have your cake and eat it too.

MSkjel, 2020-02-13:
Thank you @brett8883. Flashing 0500 worked, and all the mods are kept. Even the FCC mod. I did open my Mavic as one of the flaps holding the rear motor arms were broken. I noticed that the wires for the rear left motor were fried and also shorted together(Must be a production error, maybe the top shell squeezed the wires). I did fix that, but the error still persisted, and it goes away when flashing 0400 or 0500. Anyways, thanks for the help.

Weirdoz, 2020-02-22:
Must be something else, i have a platinum from 2018, bought used 2 days ago.
Have done the whole process also the custom 1.04.0500 at the end, all runs like it should, no problems at all.

chrischip, 2020-06-24:
I have the same problem now after the initial mod. Thank you for the advice I will try to flash 0500 and report back.

chrischip, 2020-06-26:
I flashed this firmware and check the graph in "Image Transmission Settings" and now it is below 90 (back to the original CE mode level). Looks like some of my mods (at least the FCC mod) are gone. Can I apply the jkson_fcc_mod at this stage?